aircrack-ng/rtl8812au #854: Not able to detect my AC600 Archer T2U Plus dongle

deva-gopalani, 2021-05-04:
After I run the command `sudo make && make install` i get the following error:

> install -p -m 644 88XXau.ko  /lib/modules/5.8.0-50-generic/kernel/drivers/net/wireless/
> install: cannot create regular file '/lib/modules/5.8.0-50-generic/kernel/drivers/net/wireless/88XXau.ko': Permission denied
> make: *** [Makefile:2253: install] Error 1

can someone please help me solve this?

Bee1To, 2021-05-04:
Suddenly make  && suddenly make install

Bee1To, 2021-05-04:
Sudo make && sudo make install

deva-gopalani, 2021-05-04:
thanks for the reply @Bee1To but actually my issue has been resolved

closed 2021-05-04 by deva-gopalani

Sopheaktra2050, 2022-04-20:
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.10.0-kali7-amd64/build M=/home/hacker/Desktop/rtl8812au  modules
make[1]: *** /lib/modules/5.10.0-kali7-amd64/build: No such file or directory.  Stop.
make: *** [Makefile:2244: modules] Error 2

                           How to solve this problem sir !