aircrack-ng/rtl8812au #855: No longer connecting under Linux 5.12

thomasfsteeples, 2021-05-06:
I've had great success with the driver for a while now, and I have it nstalled via DKMS on Arch Linux. However, I just upgraded from Linux 5.11.16 to 5.12.1 with `pacman -Syu` and I got the following sequence of messages,
```
==> dkms install --no-depmod -m 8812au -v 5.6.4.2_35491.20191025 -k 5.12.1-arch1-1
Error! Bad return status for module build on kernel: 5.12.1-arch1-1 (x86_64)
Consult /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/make.log for more information.
==> Warning, `dkms install --no-depmod -m 8812au -v 5.6.4.2_35491.20191025 -k 5.12.1-arch1-1' returned 10
```
I did check the make log, and from what I can remember, it was some header file problem, but because of what I did next, I'm afraid the log is now gone.

So I rebooted, removed 8812au from DKMS, pulled the latest code, and ran sudo make dkms_install. This seemingly worked fine, and the module seems to have loaded. Additionally, NetworkManager can scan and find networks, but can't authenticate. I've attached output from my journalctl ([rtl8812au.log](https://github.com/aircrack-ng/rtl8812au/files/6434954/rtl8812au.log)), but one line that stands out to me is  `nl80211: kernel reports: Authentication algorithm number required`.

Really don't know how to proceed on this, so would really appreciate any help you can provide. If you need any further details, please do let me know.

krallendertoten, 2021-05-06:
Same issue here with mostly the same process. I spent some time messing with NetworkManager and wpa_supplicant to try to narrow it down, and it definitely seems to be an issue with authentication. With both the command line and gui tools, it will continuously fail to authenticate and re-prompt for password input. 
Strangely, it also seems to cause my 5GHz network to re-appear occasionally in the list of available networks, which was something it hasn't been able to detect in a while, but loses it quickly beforr I can even enter the password for it.

NexxieIsNotACoder, 2021-05-06:
Same here, too! I thought it was just me dealing with this issue. It'd take a long time connecting to my Wi-Fi network until it gives up and disconnects. Hopefully they'll fix this ASAP.

cedricbambooza, 2021-05-07:
Got the same Problem. Used the 5.7 Version at first and when wifi stick wasn't recognized I removed it and installed the default variant. Everything via DKMS. 
Interestingly the module from the 5.7 was deactivated after Reboot after the Update on Kernel 5.12

That stuff annoys me the most. While using Linux is a wonderful thing to do, these constant unconsistencies are a pain in the ass. 
Today I update my System, whoops, my internet is completely gone and if I'm lucky nothing else is broken. 
That's the reason why my parents have to use Mac and Windows garbage, because unreliability like that is a no go area neither for normal nor for every day use. 
Makes me sad and I hope there will be a solution one day. 
Maybe a functioning way to inform the public on changes in kernel updates would be a start.

cedricbambooza, 2021-05-07:
What does work though is my WIFI stick which is supported by the kernel itself. 
So basically this couldn't be the issue.

krallendertoten, 2021-05-07:
> What does work though is my WIFI stick which is supported by the kernel itself.
> So basically this couldn't be the issue.

Not to disrupt the thread but what stick is that and would you recommend it based on speed/supported frequencies? I'd love to have a reliable backup

wwmm, 2021-05-07:
>I'd love to have a reliable backup

If you have an old router that is not used for anything and its model is supported by `openwrt` or `dd-wrt` you can transform it in a wifi client that will share internet through the cable ports. That is what I did to an old tp-link router. It is a reliable solution that will work on any operational system in computer with cable ports. But it only makes sense to do that if you have an unused router.

cedricbambooza, 2021-05-08:
So a little bit off topic but since the issue is opened and the developers can start there tinking, again...., I would say it's ok to talk about reliable alternatives. 

I'm using this stick: 

https://www.thinkpenguin.com/gnu-linux/penguin-wireless-n-usb-adapter-gnu-linux-tpe-n150usb 

It has an Atheros Chipset which is up until now the best reource I could find, since it documented with good open support on the Kernel WIFI Wiki and can do things like monitoring. 
If 150Mbit/s are enough, then that's a good solution. 
The range is moderate, but not great though. In the end doesn't have a big antenna but for that it works great

alarmfox, 2021-05-08:
Same here, not working anymore. After starting wpa_supplicant, log says: 
`nl80211: kernel reports: Authentication algorithm number required.`
Most probably something with pre-auth is off, because wpa_supplicant messages after the error says:
`CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="Vodafone-34486644" auth_failures=15 duration=229 reason=CONN_FAILED`.

opiumartdirectdesign, 2021-05-12:
can not connect wifi...
" nl80211: kernel reports: Authentication algorithm number required "
kernel : Linux 5.12.2-arch1-1
driver : rtl88xxau-aircrack-dkms-git

cedricbambooza, 2021-05-12:
yes yes, the issue persists, the solution is found, but no maintainer cares to verify it. Wonderful project, driver no broken for over 7 days. Great job!

Picea, 2021-05-12:
> yes yes, the issue persists, the solution is found, but no maintainer cares to verify it. Wonderful project, driver is now broken for nearly a week. Great job!

Got it working by editing 2 lines in a file... I do not know your OS but you can probably edit this file in your distrib package and then reinstall it manually?

cedricbambooza, 2021-05-12:
@Picea 
#857 

> Works with me, too.
> 
> `5.12.2-arch1-1 #1 SMP PREEMPT Fri, 07 May 2021 15:36:06 +0000 x86_64 GNU/Linux`

That was me dude

Picea, 2021-05-12:
:-)

wirelessrpm, 2021-05-21:
I have the latest from v5.6.4.2 and it compiles fine for 5.12.4-arch1-2, but still getting the constant auth issues. Sounds like it worked for other people. I must be missing something simple?

alxchk, 2021-05-21:
Well, I have this: 
```
 # git log | head
commit 3a64331a1c809bbbc21eca63b825970f213ec5ac
Merge: f9c8ad7 43eb4c1
Author: Christian Bremvåg <christian@aircrack-ng.org>
Date:   Tue May 4 14:27:41 2021 +0200

    Merge pull request #847 from CGarces/patch-4
    
    Remove rtl8822bu device

commit f9c8ad7192adedd9c3a943bb8d06a59dde3f9a30
```

And this doesn't work for me -- authentication failed.

cedricbambooza, 2021-05-24:
also a good alternative device: Alfa AWUS036NHA 
should work flawlessly with the kernel

gerroon, 2021-07-30:
I was using the Gordboy driver, it stopped working suddenl. I just tried the Aircrack version for the hardware, and it works. Thanks

NAP3XD, 2021-07-30:
Thanks for the update. Will try. Sent from Mail for Windows 10 From: gerroonSent: Friday, July 30, 2021 12:01 AMTo: aircrack-ng/rtl8812auCc: SubscribedSubject: Re: [aircrack-ng/rtl8812au] No longer connecting under Linux 5.12 (#855) I was using the Gordboy driver, it stopped working suddenl. I just tried the Aircrack version for the hardware, and it works. Thanks