aircrack-ng/rtl8812au #856: unable to build against 5.12.1

ZeroChaos-, 2021-05-07:
```
   ./tools/objtool/objtool orc generate  --module  --no-fp   --retpoline  --uaccess  /var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-5.6.4.2_p20210504/work/rtl8812au-059e06a51be025fde5b2bec6565540b3d9981b0b/os_dep/linux/ioctl_linux.o
  if x86_64-pc-linux-gnu-objdump -h /var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-5.6.4.2_p20210504/work/rtl8812au-059e06a51be025fde5b2bec6565540b3d9981b0b/os_dep/linux/ioctl_linux.o | grep -q __ksymtab; then x86_64-pc-linux-gnu-gcc -E -D__GENKSYMS__ -Wp,-MMD,/var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-5.6.4.2_p20210504/work/rtl8812au-059e06a51be025fde5b2bec6565540b3d9981b0b/os_dep/linux/.ioctl_linux.o.d -nostdinc -isystem /usr/lib/gcc/x86_64-pc-linux-gnu/10.2.0/include -I./arch/x86/include -I./arch/x86/include/generated  -I./include -I./arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/compiler-version.h -include ./include/linux/kconfig.h -include ./include/linux/compiler_types.h -D__KERNEL__ -fmacro-prefix-map=./= -Wall -Wundef -Werror=strict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -fshort-wchar -fno-PIE -Werror=implicit-function-declaration -Werror=implicit-int -Werror=return-type -Wno-format-security -std=gnu89 -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -fcf-protection=none -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -DCONFIG_X86_X32_ABI -Wno-sign-compare -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-branch-register -fno-jump-tables -fno-delete-null-pointer-checks -Wno-frame-address -Wno-format-truncation -Wno-format-overflow -Wno-address-of-packed-member -O2 -fno-allow-store-data-races -Wframe-larger-than=1024 -fstack-protector-strong -Wno-unused-but-set-variable -Wimplicit-fallthrough -Wno-unused-const-variable -fomit-frame-pointer -Wdeclaration-after-statement -Wvla -Wno-pointer-sign -Wno-stringop-truncation -Wno-zero-length-bounds -Wno-array-bounds -Wno-stringop-overflow -Wno-restrict -Wno-maybe-uninitialized -fno-strict-overflow -fno-stack-check -fconserve-stack -Werror=date-time -Werror=incompatible-pointer-types -Werror=designated-init -Wno-packed-not-aligned -fplugin=./scripts/gcc-plugins/structleak_plugin.so -fplugin=./scripts/gcc-plugins/randomize_layout_plugin.so -fplugin=./scripts/gcc-plugins/stackleak_plugin.so -fplugin-arg-structleak_plugin-byref-all -DSTRUCTLEAK_PLUGIN -DRANDSTRUCT_PLUGIN -DSTACKLEAK_PLUGIN -fplugin-arg-stackleak_plugin-track-min-size=100 -fplugin-arg-stackleak_plugin-arch=x86 -fno-pie -O3 -Wno-unused-variable -Wno-unused-label -Wno-unused-function -Wimplicit-fallthrough=0 -Wno-unknown-pragmas -Wno-vla -g -I/var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-5.6.4.2_p20210504/work/rtl8812au-059e06a51be025fde5b2bec6565540b3d9981b0b/include -DDRV_NAME=\"rtl88XXau\" -I/var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-5.6.4.2_p20210504/work/rtl8812au-059e06a51be025fde5b2bec6565540b3d9981b0b/platform -I/var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-5.6.4.2_p20210504/work/rtl8812au-059e06a51be025fde5b2bec6565540b3d9981b0b/hal/btc -DCONFIG_RTL8812A -DCONFIG_RTL8821A -DCONFIG_MP_INCLUDED -DCONFIG_EFUSE_CONFIG_FILE -DEFUSE_MAP_PATH=\"/system/etc/wifi/wifi_efuse_8812au.map\" -DWIFIMAC_PATH=\"/data/wifimac.txt\" -DCONFIG_LOAD_PHY_PARA_FROM_FILE -DREALTEK_CONFIG_PATH=\"/lib/firmware/\" -DCONFIG_TXPWR_BY_RATE=1 -DCONFIG_TXPWR_BY_RATE_EN=0 -DCONFIG_TXPWR_LIMIT=1 -DCONFIG_TXPWR_LIMIT_EN=0 -DCONFIG_RTW_ADAPTIVITY_EN=0 -DCONFIG_RTW_ADAPTIVITY_MODE=0 -DCONFIG_IEEE80211W -DCONFIG_RTW_80211R -DHIGH_ACTIVE_HST2DEV=0 -DCONFIG_BR_EXT '-DCONFIG_BR_EXT_BRNAME="'br0'"' -DCONFIG_WIFI_MONITOR -DCONFIG_RTW_NAPI -DCONFIG_RTW_GRO -DCONFIG_RTW_NETIF_SG -DDM_ODM_SUPPORT_TYPE=0x04 -DCONFIG_LITTLE_ENDIAN -DCONFIG_IOCTL_CFG80211 -DRTW_USE_CFG80211_STA_EVENT -I/var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-5.6.4.2_p20210504/work/rtl8812au-059e06a51be025fde5b2bec6565540b3d9981b0b/hal/phydm  -DMODULE  -DKBUILD_BASENAME='"ioctl_linux"' -DKBUILD_MODNAME='"88XXau"' -D__KBUILD_MODNAME=kmod_88XXau /var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-5.6.4.2_p20210504/work/rtl8812au-059e06a51be025fde5b2bec6565540b3d9981b0b/os_dep/linux/ioctl_linux.c | scripts/genksyms/genksyms    -r /dev/null > /var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-5.6.4.2_p20210504/work/rtl8812au-059e06a51be025fde5b2bec6565540b3d9981b0b/os_dep/linux/.tmp_ioctl_linux.ver; x86_64-pc-linux-gnu-ld -m elf_x86_64 -r -o /var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-5.6.4.2_p20210504/work/rtl8812au-059e06a51be025fde5b2bec6565540b3d9981b0b/os_dep/linux/.tmp_ioctl_linux.o /var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-5.6.4.2_p20210504/work/rtl8812au-059e06a51be025fde5b2bec6565540b3d9981b0b/os_dep/linux/ioctl_linux.o -T /var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-5.6.4.2_p20210504/work/rtl8812au-059e06a51be025fde5b2bec6565540b3d9981b0b/os_dep/linux/.tmp_ioctl_linux.ver; mv -f /var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-5.6.4.2_p20210504/work/rtl8812au-059e06a51be025fde5b2bec6565540b3d9981b0b/os_dep/linux/.tmp_ioctl_linux.o /var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-5.6.4.2_p20210504/work/rtl8812au-059e06a51be025fde5b2bec6565540b3d9981b0b/os_dep/linux/ioctl_linux.o; rm -f /var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-5.6.4.2_p20210504/work/rtl8812au-059e06a51be025fde5b2bec6565540b3d9981b0b/os_dep/linux/.tmp_ioctl_linux.ver; fi
make[1]: *** [Makefile:1851: /var/tmp/portage/net-wireless/rtl8812au_aircrack-ng-5.6.4.2_p20210504/work/rtl8812au-059e06a51be025fde5b2bec6565540b3d9981b0b] Error 2
make[1]: Leaving directory '/usr/src/linux-5.12.1-pentoo'
make: *** [Makefile:2244: modules] Error 2
```

CGarces, 2021-05-07:
Fixed on e7e83f2593c9e67e3ee50d032f1ad39fe47ea81d
Please build using the last version of the source code.

ZeroChaos-, 2021-05-07:
Whoops, thought I was on the latest and very much wasn't.  Sorry for the noise, I can confirm it builds.  Thanks!

closed 2021-05-07 by ZeroChaos-