aircrack-ng/rtl8812au #861: rtl8812au Compile Error "rtw_select_queue"

jtome7, 2021-05-16:
May I please ask if there's an easy way to get around this compile error?
Using CentOS  ( which is like Fedora & Red Hat ) kernel 4.18

/var/lib/dkms/rtl8812au/5.2.20.2/build/os_dep/linux/os_intfs.c:1393:22: error: initialization of ‘u16 (*)(struct net_device *, struct sk_buff *, struct net_device *, u16 (*)(struct net_device *, struct sk_buff *, struct net_device *))’ {aka ‘short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *, short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *))’} from incompatible pointer type ‘u16 (*)(struct net_device *, struct sk_buff *, void *, u16 (*)(struct net_device *, struct sk_buff *, struct net_device *))’ {aka ‘short unsigned int (*)(struct net_device *, struct sk_buff *, void *, short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *))’} [-Werror=incompatible-pointer-types]
  .ndo_select_queue = rtw_select_queue,
                      ^~~~~~~~~~~~~~~~
/var/lib/dkms/rtl8812au/5.2.20.2/build/os_dep/linux/os_intfs.c:1393:22: note: (near initialization for ‘rtw_netdev_ops.ndo_select_queue’)

CGarces, 2021-05-18:
CentOS should work after #599 on branch v5.6.4.2
¿Are you using a different branch?

jtome7, 2021-05-18:
Hello, Carlos!

There are millions of Linux users world-wide on the distribution family of
Red Hat / CentOS / Fedora Linux, so thanks for adding support for Realtek
8812 & 8814 chipset for those millions of users!

The title says rtl8812au ... does it also work for rtl8814au that is found
in quite a number of wifi adapter products like this one:

TP-Link AC1900 Archer T9UH

[image: TP-Link-Archer-T9UH-WIFI-Adapter.jpg]

*Thanks!**Joel*

*************************** Joel Tome *

*jtome7 @ gmail.com <http://gmail.com>****************************

CGarces, 2021-05-18:
Support for the rtl8814au chipset was removed from this driver. Use https://github.com/aircrack-ng/rtl8814au/

jtome7, 2021-05-18:
Thanks, I'll check it out.

*Thanks!**Joel*

*************************** Joel Tome *

*jtome7 @ gmail.com <http://gmail.com>****************************

jtome7, 2021-05-19:
Just to let folks know ... I tried several 5.x kernels from the ElRepo ... nothing worked on my CentOS 8.3 with my TP-Link AC1900 Archer t9UH wifi adapter with Realtek 8814 chipset.  Thanks for continuing suggestions as new updates become available.  Appreciate your efforts!

lepp9922, 2021-06-09:
I can also confirm the same error, compiling fails on the latest AlmaLinux 8 (RHEL/CentOS 8). Kernel 4.18

jtome7, 2021-06-11:
The crazy thing is that just before the 8814au driver was separated from this rtl8812au branch,  the TP-Link AC1900 Archer t9UH wifi adapter with Realtek 8814 chipset WORKED!  I wish there were a way to get that previous compilation back that contained both 8812 & 8814.

CGarces, 2021-06-11:
I have fixed the compilation
https://github.com/aircrack-ng/rtl8812au/compare/v5.6.4.2...CGarces:centos
Tested on CentOS 8 VM, not on real hardware.

Rondore, 2021-06-28:
The patch by @CGarces allowed me to compile this driver for Rocky Linux 8.4 running kernel 4.18.0-305.3.1.el8_4.x86_64. I ran this test on real hardware and my WiFi is now working great. Thanks!

kimocoder, 2021-12-12:
rtl8814au support is added back and fixes has been pushed. Update and build/install.

If problem persists, open this issue report again :+1:

closed 2021-12-12 by kimocoder

fs30000, 2023-01-13:
> I have fixed the compilation [v5.6.4.2...CGarces:centos](https://github.com/aircrack-ng/rtl8812au/compare/v5.6.4.2...CGarces:centos) Tested on CentOS 8 VM, not on real hardware.

CGarces, i have Rocky 8.7, it fails to compile as well. I don't know much about git, it there a way to clone your repo with all the changes for Centos/Rocly/Redhat?