aircrack-ng/rtl8812au #863: Cannot install on latest version of kali linux | Error 1

Aholicknight, 2021-05-22:
[image]

Every time I try to install this on the latest version of kali linux this error pops up, does anyone know how to fix it? Thanks.

LinkerOK, 2021-05-22:
Use the latest branch 5.7.0

LinkerOK, 2021-05-22:
> 
> 
> Use the latest branch 5.7.0

git clone -b v5.7.0 https://github.com/aircrack-ng/rtl8812au.git

yelsanya, 2021-05-31:
Kinda not sure about your version being the latest one, consider running this:
`sudo apt update && sudo apt upgrade`
Also for your issue it seems like you need this
`sudo apt-get install linux-headers-$(uname -r)`
Also do not forget to reboot your machine after that

closed 2021-12-12 by kimocoder