aircrack-ng/rtl8812au #864: build failure in fedora 33 for kernel 5.12.5-200.fc33.x86_64

karlkleinpaste, 2021-05-23:
After ensuring that I'm up to date...
```
git status
On branch v5.7.0
Your branch is up to date with 'origin/v5.7.0'.
```
where HEAD is:
```
commit 1c9d034b20aa5c15dbf5bb5dfcb83346a692f827 (HEAD -> v5.7.0, origin/v5.7.0)
```
Build fails for latest 5.12.5 kernel:
```
# dkms build 8812au/5.9.3.2_37279.20201012 -k 5.12.5-200.fc33.x86_64

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' -j12 KVER=5.12.5-200.fc33.x86_64 KSRC=/lib/modules/5.12.5-200.fc33.x86_64/build.....(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.12.5-200.fc33.x86_64 (x86_64)
Consult /var/lib/dkms/8812au/5.9.3.2_37279.20201012/build/make.log for more information.
```
Relevant portions of that log:

```
/var/lib/dkms/8812au/5.9.3.2_37279.20201012/build/os_dep/linux/recv_linux.c: In function ‘napi_recv’:
/var/lib/dkms/8812au/5.9.3.2_37279.20201012/build/os_dep/linux/recv_linux.c:358:55: error: ‘GRO_DROP’ undeclared (first use in this function)
  358 |    if (rtw_napi_gro_receive(&padapter->napi, pskb) != GRO_DROP)
      |                                                       ^~~~~~~~
make[2]: *** [scripts/Makefile.build:271: /var/lib/dkms/8812au/5.9.3.2_37279.20201012/build/os_dep/linux/recv_linux.o] Error 1
/var/lib/dkms/8812au/5.9.3.2_37279.20201012/build/hal/hal_com.c: In function ‘rtw_hal_set_8812a_vendor_ie’:
/var/lib/dkms/8812au/5.9.3.2_37279.20201012/build/hal/hal_com.c:15445:2: warning: ISO C90 forbids variable length array ‘vendor_info’ [-Wvla]
15445 |  unsigned char vendor_info[vender_len];
      |  ^~~~~~~~
make[1]: *** [Makefile:1852: /var/lib/dkms/8812au/5.9.3.2_37279.20201012/build] Error 2
```

CGarces, 2021-05-23:
You need to apply #857 to v5.7.0, but consider use the default branch

karlkleinpaste, 2021-05-23:
thanx, done.

closed 2021-05-23 by karlkleinpaste