aircrack-ng/rtl8812au #866: ubuntu 21.04 - chipset 8811au - Kernel 5.11.x. It worked 3 days and stopped

neuberfran, 2021-05-24:
Hi,

I've been trying to get my wifi dongle (8811au) back on ubuntu 21.04 for 3 days.

**Can anyone help**

neuberfran@ubuntu2104:~$ **nmcli con**
NAME UUID TYPE DEVICE
Conexão cabeada 1 4f0a34b8-d394-3914-b110-164598dfb94a ethernet enp3s0
pablo 4a2dca15-84ed-4fba-837a-a732957991f1 wifi --

neuberfran@ubuntu2104:~$ **nmcli con up pablo**
Erro: Ativação da conexão falhou: No suitable device found for this connection (device lo not available because device is strictly unmanaged).

neuberfran@ubuntu2104:~$ **iwconfig**
lo no wireless extensions.
enp3s0 no wireless extensions.
enxd03745db188a unassociated Nickname:"WIFI@REALTEK"
Mode:Auto Frequency=2.412 GHz Access Point: Not-Associated
Sensitivity:0/0
Retry:off RTS thr:off Fragment thr:off
Power Management:off
Link Quality=0/100 Signal level=0 dBm Noise level=0 dBm
Rx invalid nwid:0 Rx invalid crypt:0 Rx invalid frag:0
Tx excessive retries:0 Invalid misc:0 Missed beacon:0

neuberfran@ubuntu2104:~$ **route -n**
Tabela de Roteamento IP do Kernel
Destino Roteador MáscaraGen. Opções Métrica Ref Uso Iface
0.0.0.0 192.168.1.1 0.0.0.0 UG 100 0 0 enp3s0
169.254.0.0 0.0.0.0 255.255.0.0 U 1000 0 0 enp3s0
192.168.1.0 0.0.0.0 255.255.255.0 U 100 0 0 enp3s0

antimoroma, 2021-08-12:
Same problem here work fine for about two hour , then after a reboot it stop working.
Feel free to ask me more info if needed

neuberfran, 2021-08-12:
@antimoroma try this: https://askubuntu.com/questions/1341262/dongle-wifi-tp-link-ac600-archer-t2u-8811au-not-working-in-ubuntu-21-04

I don't have more ubuntu-21.04. I have 20.10 and 20.04. In these two it's been working OK.

antimoroma, 2021-08-13:
Thank you for your update, it seems that in my case the problem was related to my mobile hot spot, after restarting the hot spot it seems to work fine with a really good speed :). Sorry, i didn't think about this restart before writing here :)

wawrzek, 2021-09-26:
@antimoroma I suggest to close the issue then.

antimoroma, 2021-09-27:
Uhm i think  @neuberfran can close the issue as far as I see I can see only a comment button

closed 2025-06-25 by neuberfran