aircrack-ng/rtl8812au #869: Wifi error connecting using kernel >= 5.12.5 (tested on new install and existing installation of driver)

CharlesHagerman, 2021-05-26:
The Realtek 8812au driver was installed and working perfectly until 5.11.20, upgraded kernel to 5.12.5 wireless card indicates it's working but does not connect

Uninstalled the driver, the re-installed driver will not connect to wireless using kernel 5.12.5

downgraded kernel to 5.11.20, wireless networks card works perfectly
upgraded to kernel to 5.12.5, wireless network card will not connect but shows

Output to show driver loaded on both kernels...

sudo dkms status
8812au, 5.6.4.2_35491.20191025, 5.11.20-300.fc34.x86_64, x86_64: installed
8812au, 5.6.4.2_35491.20191025, 5.12.5-300.fc34.x86_64, x86_64: installed

CharlesHagerman, 2021-05-26:
> Duplicate of #855
> Please appy patch #857

Thank-you for the patch, wireless works perfectly on all versions of the kernel indicated below that we individually tested after the patch was installed

dkms status
8812au, 5.6.4.2_35491.20191025, 5.11.20-300.fc34.x86_64, x86_64: installed
8812au, 5.6.4.2_35491.20191025, 5.12.5-300.fc34.x86_64, x86_64: installed
8812au, 5.6.4.2_35491.20191025, 5.12.6-300.fc34.x86_64, x86_64: installed

Much appreciate the work done to keep the drivers updated for use
Have a good day!

rohan-0100010110010100, 2021-06-24:
> Duplicate of #855
> Please appy patch #857

Not working on Kernel 5.12.11, OS - Fedora 34

The patch was working till Kernel 5.12.6 and suddenly stopped after upgrade to 5.12.11.

I think I have to shift to T4U plus adapters.