aircrack-ng/rtl8812au #873: Build failed with ‘struct cfg80211_ops’ has no member named ‘mgmt_frame_register’ on CentOS 8

lightingghost, 2021-06-12:
Hi,

I am trying to build from source but got the following error when running `sudo make dkms_install`:

```
/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/os_dep/linux/ioctl_cfg80211.c:10191:3: error: ‘struct cfg80211_ops’ has no member named ‘mgmt_frame_register’
  .mgmt_frame_register = cfg80211_rtw_mgmt_frame_register,
   ^~~~~~~~~~~~~~~~~~~
/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/os_dep/linux/ioctl_cfg80211.c:10191:25: error: initialization of ‘int (*)(struct wiphy *, struct wireless_dev *, u64)’ {aka ‘int (*)(struct wiphy *, struct wireless_dev *, long long unsigned int)’} from incompatible pointer type ‘void (*)(struct wiphy *, struct wireless_dev *, u16,  bool)’ {aka ‘void (*)(struct wiphy *, struct wireless_dev *, short unsigned int,  _Bool)’} [-Werror=incompatible-pointer-types]
  .mgmt_frame_register = cfg80211_rtw_mgmt_frame_register,
                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/os_dep/linux/ioctl_cfg80211.c:10191:25: note: (near initialization for ‘rtw_cfg80211_ops.mgmt_tx_cancel_wait’)
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/hal/hal_dm.o
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:315: /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/os_dep/linux/ioctl_cfg80211.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [Makefile:1563: _module_/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build] Error 2
make[2]: Leaving directory '/usr/src/kernels/4.18.0-305.3.1.el8.x86_64'
make[1]: *** [Makefile:2244: modules] Error 2
make[1]: Leaving directory '/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build'
```

Any help will be appreciated

CGarces, 2021-06-12:
Please check isssue #861
And this patch
https://github.com/aircrack-ng/rtl8812au/issues/861#issuecomment-859794427

lightingghost, 2021-06-12:
> Please check isssue #861
> And this patch
> [#861 (comment)](https://github.com/aircrack-ng/rtl8812au/issues/861#issuecomment-859794427)

Thank you for your help! This works for me!

closed 2021-06-12 by lightingghost