aircrack-ng/rtl8812au #878: Realtek WiFi USB Adapter does not work.

palmengladosTV, 2021-08-17:
When typing `lsusb` the WiFi-Stick appears as following:
_Bus 003 Device 003: ID 0bda:b812 Realtek Semiconductor Corp. USB3.0 802.11ac 1200M Adapter_

But after installing the driver as described, the USB Stick still only appears under `lsusb` and there is no sign of any wifi device under `ip link show`. And the driver should be the right one for that modell. I have currently the 4.3.0-81 Linux kernel on my system.

Help would be much appreciated. :) If you need any more logs or something else just let me know ^^

wawrzek, 2021-09-26:
Is the module loaded? What does `lsmod` show?

palmengladosTV, 2021-11-08:
Resolved. Works flawlessly for over a month now.

closed 2021-11-08 by palmengladosTV