aircrack-ng/rtl8812au #879: Issue with OpenSUSE Leap 15.3 - DKMS installation failed

sh0m3r, 2021-08-17:
In OpenSUSE Leap 15.3 I ran the following commands: (for a TP-Link AC600 T2U Plus WIFI Adapter)

`$ sudo zypper install kernel-source`
`$ git clone https://github.com/aircrack-ng/rtl8812au.git`
`$ cd rtl8812au/`
`$ sudo make dkms_install`

I am getting the following error message:

```
mkdir -p /usr/src/8812au-5.6.4.2_35491.20191025
cp -r * /usr/src/8812au-5.6.4.2_35491.20191025
dkms add -m 8812au -v 5.6.4.2_35491.20191025

Creating symlink /var/lib/dkms/8812au/5.6.4.2_35491.20191025/source ->
                 /usr/src/8812au-5.6.4.2_35491.20191025

DKMS: add completed.
dkms build -m 8812au -v 5.6.4.2_35491.20191025

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' -j4 KVER=5.3.18-59.19-default KSRC=/lib/modules/5.3.18-59.19-default/build...........(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.3.18-59.19-default (x86_64)
Consult /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/make.log for more information.
make: *** [Makefile:2304: dkms_install] Error 10

```

Please help!

xrucka, 2021-08-20:
Hello, I'm also running Leap 15.3, we might have the same issue. Can you please paste the relevant part of /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/make.log ?

sh0m3r, 2021-08-20:
Yes. I pasted part of the log file at the bottom.

There is a thread about it in the openSUSE forum - worthwhile checking out.
[https://forums.opensuse.org/showthread.php/558419-TP-Link-AC600-Driver-installation-failed](url)
Let me know If it helped.

`/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/os_dep/linux/ioctl_cfg80211.c:10191:3: error: ‘struct cfg80211_ops’ has no member named ‘mgmt_frame_register’
  .mgmt_frame_register = cfg80211_rtw_mgmt_frame_register,
   ^~~~~~~~~~~~~~~~~~~
/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/os_dep/linux/ioctl_cfg80211.c:10191:25: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .mgmt_frame_register = cfg80211_rtw_mgmt_frame_register,
                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/os_dep/linux/ioctl_cfg80211.c:10191:25: note: (near initialization for ‘rtw_cfg80211_ops.mgmt_tx_cancel_wait’)
cc1: some warnings being treated as errors
make[4]: *** [/usr/src/linux-5.3.18-59.19/scripts/Makefile.build:287: /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/os_dep/linux/ioctl_cfg80211.o] Error 1
make[4]: *** Waiting for unfinished jobs....
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/os_dep/linux/rtw_android.o
make[3]: *** [/usr/src/linux-5.3.18-59.19/Makefile:1675: _module_/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build] Error 2
make[2]: *** [../../../linux-5.3.18-59.19/Makefile:179: sub-make] Error 2
make[2]: Leaving directory '/usr/src/linux-5.3.18-59.19-obj/x86_64/default'
make[1]: *** [Makefile:2244: modules] Error 2
make[1]: Leaving directory '/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build'
`

kimocoder, 2021-12-12:
Updates has been pushed. Update and build/install.

If the problem persists, open the issue report again :+1:

closed 2021-12-12 by kimocoder