aircrack-ng/rtl8812au #885: Upstreaming to Linus' kernel?

zboszor, 2021-09-11:
It would be nicer if support for this USB Realtek chip was in the mainline kernel. Thanks in advance.

stkw0, 2022-06-03:
I'm also interested about the reasons for which this is not upstream

CGarces, 2022-06-03:
It's a huge amount of work.
A lot of dead code from others S.O. (windows, android, ect), source code from other devices, custom code that must be replaced with the equivalence in the linux kernel...

A good point to start is replicate the 805 commits from this driver that is now on linux to untherstand the process from one rtl* driver to linux /drivers/staging/
https://github.com/hadess/rtl8723bs
After that, replicate the commits from
https://github.com/torvalds/linux/tree/master/drivers/staging/rtl8723bs
and also take a look at
https://github.com/torvalds/linux/tree/master/drivers/staging/r8188eu

And that is only the easy part... after that work maybe you can land to linux /drivers/staging/ when the difficult part starts, because the driver must be maintained.

stkw0, 2022-06-03:
I see. Thank you very much!