aircrack-ng/rtl8812au #892: DKMS install fails on Debian running Kernel v 5.14

AmNe5iA, 2021-09-28:
sudo dkms install 8812au/5.9.3.2_37279.20201012 -k 5.14.0-1-amd64/x86_64

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' -j4 KVER=5.14.0-1-amd64 KSRC=/lib/modules/5.14.0-1-amd64/build........(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.14.0-1-amd64 (x86_64)
Consult /var/lib/dkms/8812au/5.9.3.2_37279.20201012/build/make.log for more information.
[make.log](https://github.com/aircrack-ng/rtl8812au/files/7244830/make.log)

kimocoder, 2021-12-12:
This is not related to this driver.
812au/5.9.3.2_37279.20201012 says you trying v5.9.3.2.. we're using v5.6.4.2.

closed 2021-12-12 by kimocoder

AmNe5iA, 2021-12-12:
5.9.3.2 is the version number that reports when downloading the 5.7.0 branch of this driver (or at least it was when I reported this)

kimocoder, 2021-12-12:
Don't use that branch, not maintained.