aircrack-ng/rtl8812au #895: AWUS036ACM Extremely poor range

deanfourie1, 2021-10-03:
I just purchased a AWUS036ACM, and im getting extremely poor rage. Can anyone suggest the best driver to use with this adapter? Is it the 8812au driver?

I cannot scan networks past the range of my own house.

Thanks in advance.

kimboslice99, 2022-11-11:
> Here is a link that shows performance testing regarding the statement above:
> 
> https://github.com/morrownr/USB-WiFi/blob/main/Performance_Comparison.md

Don't mean to necro, but what happened to this perf comparison

morrownr, 2022-11-11:
It is in the home folder at this location:

https://github.com/morrownr/USB-WiFi

If interested in a speed test, go to the same repo and look for issue 139.

Regards