aircrack-ng/rtl8812au #904: DKMS Install Error #1 on 5.10.0-kali3-amd64

Aholicknight, 2021-11-01:
Trying to install this network adapter on 5.10.0-kali3-amd64 but this is the message i get attempting to do so:
```
mkdir -p /usr/src/8812au-5.6.4.2_35491.20191025
cp -r * /usr/src/8812au-5.6.4.2_35491.20191025
dkms add -m 8812au -v 5.6.4.2_35491.20191025
Creating symlink /var/lib/dkms/8812au/5.6.4.2_35491.20191025/source -> /usr/src/8812au-5.6.4.2_35491.20191025
dkms build -m 8812au -v 5.6.4.2_35491.20191025
Error! Your kernel headers for kernel 5.10.0-kali3-amd64 cannot be found at /lib/modules/5.10.0-kali3-amd64/build or /lib/modules/5.10.0-kali3-amd64/source.
You can use the --kernelsourcedir option to tell DKMS where it's located.
make: *** [Makefile:2304: dkms_install] Error 1
```

anyone know how to fix this? thanks

vishytk, 2021-11-04:
kernel header files are missing, on Fedora the package that provides the required files is kernel-devel

wifislax-ng, 2021-11-04:
> kernel header files are missing, on Fedora the package that provides the required files is kernel-devel

Not true , the problem is a removed file from new 5.15.x kernel.
I cant build on slackware64-current and i have headers , but this one was removed on 5.15
**/tmp/build/rtl88XXau/rtl8812au/core/rtw_br_ext.c:20:18: error fatal: net/ipx.h: No such file or directory**

kimocoder, 2021-12-12:
Pushed fixes for kernel v5.15-5+ and minor DKMS fix.

However, @wifislax-ng this is another issue related outside Kali Linux.. let me see if I may fix that too

kimocoder, 2021-12-12:
Fixes has been pushed. Update and build/install driver again.

If problem persists, open a new issue report. :+1:

closed 2021-12-12 by kimocoder