aircrack-ng/rtl8812au #907: installation on lxc Kali gives  `make[1]: *** /lib/modules/5.4.0-89-generic/build: No such file or directory.  Stop.`

SpIIIII, 2021-11-06:
Honestly don't know if it an issue just trying to get help and some understanding. And wondering why no one seemingly encounter it (can't find related issue).
   I tried to install this driver on freshly installed kali system (kali run through lxc from Ubuntu system). After run `make` I've got
```
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.4.0-89-generic/build M=/home/kali/downloads/rtl8812au  modules
make[1]: *** /lib/modules/5.4.0-89-generic/build: No such file or directory.  Stop.
```
   It's a bit strange because in `/lib/modules` I have only `5.14.0-kali2-amd64` folder, and what's even more interesting for me, that when I installed this driver on main Ubuntu it works fine and select the same `/lib/modules/5.4.0-89-generic/build` destination (which obv works fine).
   Cos I have nothing to lose on this system (kali) I manually changed `/lib/modules/5.4.0-89-generic/build` to `/lib/modules/5.14.0-kali2-amd64/build` (didn't expect that will work just interesting what's will happen). `make` went seemingly well, but `make install` tried to reach `/lib/modules/5.4.0-89-generic/kernel/drivers/net/wireless`. After change that to `/lib/modules/5.14.0-kali2-amd64/kernel/drivers/net/wireless` (and creating missing `kernel/drivers/net/wireless` folders) nothing (expectedly) worked.
   I'm doing this following  "David Bombal" 's youtube channel video instruction and pretty surprised that no one in comments face this issue while I experience it on 2 Ubuntu and 2 kali systems.

kimocoder, 2021-11-06:
You're missing the kernel-headers. 
"apt-cache search linux-headers" to find correct one, or just read the readme

closed 2021-11-06 by kimocoder

SpIIIII, 2021-11-06:
> You're missing the kernel-headers. "apt-cache search linux-headers" to find correct one, or just read the readme

Thx I read Readme, either I completely  misunderstand my problem and your answer or you didn't read my comment due to it's verbosity
I tryed `sudo apt-get install linux-headers-$(uname -r)` but I've got:

`E: Unable to locate package linux-headers-`**`5.4.0-89-generic`**
`E: Couldn't find any package by glob 'linux-headers-`**`5.4.0-89-generic`**`'`
`E: Couldn't find any package by regex 'linux-headers-`**`5.4.0-89-generic`**`'`

And `sudo apt-get install linux-headers-generic` gives:
```
Note, selecting 'linux-headers-amd64' instead of 'linux-headers-generic'
linux-headers-amd64 is already the newest version (5.14.9-2kali1).
```
Also `dpkg-query -s linux-headers-amd64` return this:

```
Package: linux-headers-amd64
Status: install ok installed
Priority: optional
Section: kernel
Installed-Size: 10
Maintainer: Kali Developers <devel@kali.org>
Architecture: amd64
Source: linux
Version: 5.14.9-2kali1
Provides: linux-headers-generic
Depends: linux-headers-5.14.0-kali2-amd64 (= 5.14.9-2kali1)
Description: Header files for Linux amd64 configuration (meta-package)
 This package depends on the architecture-specific header files for the
 latest Linux kernel amd64 configuration.
Homepage: https://www.kernel.org/
```

Btw isn't `/lib/modules/`**`5.14.0-kali2-amd64`**`/` is already installed headers
Also is it ok that `uname -r` returns me **`5.4.0-89-generic`** which completely identical result with my **host** Ubuntu system (**I run Kali in lxc)**

Sorry for bothering again, but I'll further trying to resolve my problem

gh4reeb, 2021-11-24:
exactly same issue @spllllll

gh4reeb, 2021-11-24:
hey right now solve my problem by downloading the **linux-headers-5.14.0-kali2-amd64** and **linux-headers-5.14.0-kali2-common** from [](http://http.kali.org/kali/pool/main/l/linux/) 
then installing
```
dpkg -i linux-headers-5.14.0-kali2-common
dpkg -i linux-headers-5.14.0-kali2-amd64
```

SpIIIII, 2021-12-12:
> 
thx, I tried that, but nothing works for me (
firstly I can't find exactly **linux-headers-5.14.0-kali2-amd64** and **linux-headers-5.14.0-kali2-common**  (maybe I don't know some common thing about downloading from http://http.kali.org/kali/pool/main/l/linux/ )
Assuming that I need this not literally, I loaded and installed **linux-headers-5.14.0-kali2-amd64_5.14.9-2kali1_amd64.deb** and **linux-headers-5.14.0-kali2-common_5.14.9-2kali1_all.deb**. Which not lead me anywhere (
in `/lib/modules/` still have only one **`5.14.0-kali2-amd64/`**
`make` command gives me exactly same error (like in first attempt)

Appreciate that you read my question and comment, and offered help.

gh4reeb, 2021-12-13:
Dear wil You you show me. The output of dpkg -i linux-headers-5.14.0-kali2-common
dpkg -i linux-headers-5.14.0-kali2
Actually my problem was solved by this. Maybe it is colide with other headers. I will be really happy to solve your problem if it is not solved yet.