aircrack-ng/rtl8812au #910: Broken on kernel 5.15.2

JoshuaJakowlew, 2021-11-13:
Hi, today I updated the kernel (Arch linux, 5.14.16 -> 5.15.2 via pacman -Syu). After reboot driver did not run. In logs I understood, that it failed to build. 

Here is the log entry with error:
```
[2021-11-13T02:49:18+0300] [ALPM-SCRIPTLET] ==> dkms install --no-depmod -m rtl88xxau -v r1164.b8167e6 -k 5.15.2-arch1-1
[2021-11-13T02:49:24+0300] [ALPM-SCRIPTLET] Error! Bad return status for module build on kernel: 5.15.2-arch1-1 (x86_64)
[2021-11-13T02:49:24+0300] [ALPM-SCRIPTLET] Consult /var/lib/dkms/rtl88xxau/r1164.b8167e6/build/make.log for more information.
[2021-11-13T02:49:24+0300] [ALPM-SCRIPTLET] ==> Warning, `dkms install --no-depmod -m rtl88xxau -v r1164.b8167e6 -k 5.15.2-arch1-1' returned 10
```

lolmokha, 2021-11-13:
Try pulling the latest changes from the repo. That worked for me. If that doesn't work try recloning the repository and build it again.

theinvisiblewizard1674, 2021-11-16:
> Try pulling the latest changes from the repo. That worked for me. If that doesn't work try recloning the repository and build it again.

Hello, newbie here, could you please explain it a little? I can't understand what you're saying.

lolmokha, 2021-11-16:
How did you download the drivers?

CGarces, 2021-11-16:
> Here: https://aur.archlinux.org/packages/8188eu-aircrack-dkms-git/

So comment your issue at https://aur.archlinux.org/packages/8188eu-aircrack-dkms-git/
Also the information at /var/lib/dkms/rtl88xxau/r1164.b8167e6/build/make.log  can be usefully to solve your issue.

lolmokha, 2021-11-16:
> Here: https://aur.archlinux.org/packages/8188eu-aircrack-dkms-git/

right so here's what you do:
- git clone https://github.com/aircrack-ng/rtl8812au.git in your terminal
-  sudo pacman -S dkms if you don't have it installed
- sudo make dkms_install

theinvisiblewizard1674, 2021-11-18:
Yes, I did install it using a package manager at first.

lolmokha, 2021-11-18:
Then try upgrading packages using the package manager(ex, if you're using yay, just type in yay -Syyu). But  if you're not, do what I said ppreviously

mindflayer, 2021-11-20:
Same for Ubuntu:
```
~/r/rtl8812au (v5.6.4.2) $ cat /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/make.log
DKMS make.log for 8812au-5.6.4.2_35491.20191025 for kernel 5.15.0-3.1-liquorix-amd64 (x86_64)
dom 21 nov 2021, 00:06:54, CET
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.15.0-3.1-liquorix-amd64/build M=/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build  modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-3.1-liquorix-amd64'
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_mlme.o
/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_mlme.c: In function ‘rtw_drv_scan_by_self’:
/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_mlme.c:3196:17: warning: this ‘else’ clause does not guard... [-Wmisleading-indentation]
 3196 |                 else
      |                 ^~~~
/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_mlme.c:3199:25: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘else’
 3199 |                         goto exit;
      |                         ^~~~
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_pwrctrl.o
/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_wlan_util.c: In function ‘rtw_parse_sta_vendor_ie_8812’:
/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_wlan_util.c:3226:25: warning: this ‘for’ clause does not guard... [-Wmisleading-indentation]
 3226 |                         for(i=0; i<9;i++)
      |                         ^~~
In file included from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/include/drv_types.h:51,
                 from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_wlan_util.c:17:
/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/include/rtw_debug.h:52:33: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘for’
   52 |         #define RTW_INFO(x,...) do {} while (0)
      |                                 ^~
/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_wlan_util.c:3228:33: note: in expansion of macro ‘RTW_INFO’
 3228 |                                 RTW_INFO("\n");
      |                                 ^~~~~~~~
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_chplan.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_sta_mgt.o
/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_rf.c: In function ‘dump_txpwr_lmt’:
/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_rf.c:775:9: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
  775 |         if (IS_HARDWARE_TYPE_JAGUAR_AND_JAGUAR2(adapter))
      |         ^~
In file included from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/include/drv_types.h:51,
                 from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_rf.c:17:
/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/include/rtw_debug.h:220:9: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘if’
  220 |         do {\
      |         ^~
/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_rf.c:777:17: note: in expansion of macro ‘RTW_PRINT_SEL’
  777 |                 RTW_PRINT_SEL(sel, "txpwr_lmt_5g_20_40_ref:0x%02x\n", rfctl->txpwr_lmt_5g_20_40_ref);
      |                 ^~~~~~~~~~~~~
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/mesh/rtw_mesh.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_rson.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_br_ext.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_sreset.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_btcoex_wifionly.o
/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_br_ext.c:20:18: fatal error: net/ipx.h: No such file or directory
   20 |         #include <net/ipx.h>
      |                  ^~~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:277: /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_br_ext.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:1875: /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-3.1-liquorix-amd64'
make: *** [Makefile:2244: modules] Error 2
~/r/rtl8812au (v5.6.4.2) $ git pull
Already up-to-date.
```

CGarces, 2021-11-20:
@mindflayer your issue was fixed on #898

mindflayer, 2021-11-22:
I had to "sudo make dkms_remove" and "sudo make dkms_install" for forcing the compiling after I pulled the last changes. Thanks!

JoshuaJakowlew, 2021-11-25:
> > Here: https://aur.archlinux.org/packages/8188eu-aircrack-dkms-git/
> 
> right so here's what you do:
> 
>     * git clone https://github.com/aircrack-ng/rtl8812au.git in your terminal
> 
>     * sudo pacman -S dkms if you don't have it installed
> 
>     * sudo make dkms_install
> 
> 
> EDIT: firstly, did you install it using a package manager?

dkms_install gave me some errors about missing dkms.conf. Then I manually removed previous driver installed via pacman and tried again. It compiles, tries to install and fails again with the same error after installation.

Nevertheless, after reboot everything works

theinvisiblewizard1674, 2021-11-30:
` DKMS make.log for 8188eu-5.3.9.r145.6146193 for kernel 5.15.5-arch1-1 (x86_64)
Tue Nov 30 03:42:29 PM +0545 2021
cut: /etc/redhat-release: No such file or directory
/bin/sh: line 1: bc: command not found
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.15.5-arch1-1/build M=/var/lib/dkms/8188eu/5.3.9.r145.6146193/build  modules
make[1]: Entering directory '/usr/lib/modules/5.15.5-arch1-1/build'
cut: /etc/redhat-release: No such file or directory
/bin/sh: line 1: bc: command not found
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/mesh/rtw_mesh.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_rson.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_br_ext.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_sreset.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_btcoex_wifionly.o
/var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_br_ext.c:20:18: fatal error: net/ipx.h: No such file or directory
   20 |         #include <net/ipx.h>
      |                  ^~~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:277: /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_br_ext.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:1868: /var/lib/dkms/8188eu/5.3.9.r145.6146193/build] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.15.5-arch1-1/build'
make: *** [Makefile:2062: modules] Error 2 `

i don't know what to make of this

theinvisiblewizard1674, 2021-11-30:
`❯ sudo cat make.log
DKMS make.log for 8188eu-5.3.9.r145.6146193 for kernel 5.15.5-arch1-1 (x86_64)
Tue Nov 30 03:42:29 PM +0545 2021
cut: /etc/redhat-release: No such file or directory
/bin/sh: line 1: bc: command not found
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.15.5-arch1-1/build M=/var/lib/dkms/8188eu/5.3.9.r145.6146193/build  modules
make[1]: Entering directory '/usr/lib/modules/5.15.5-arch1-1/build'
cut: /etc/redhat-release: No such file or directory
/bin/sh: line 1: bc: command not found
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/mesh/rtw_mesh.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_rson.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_br_ext.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_sreset.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_btcoex_wifionly.o
/var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_br_ext.c:20:18: fatal error: net/ipx.h: No such file or directory
   20 |         #include <net/ipx.h>
      |                  ^~~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:277: /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_br_ext.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:1868: /var/lib/dkms/8188eu/5.3.9.r145.6146193/build] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.15.5-arch1-1/build'
make: *** [Makefile:2062: modules] Error 2
`
_____________________________________________________________________________________________________

I don't know what to make of this.

theinvisiblewizard1674, 2021-11-30:
❯ sudo cat make.log
DKMS make.log for 8188eu-5.3.9.r145.6146193 for kernel 5.15.5-arch1-1 (x86_64)
Tue Nov 30 03:42:29 PM +0545 2021
cut: /etc/redhat-release: No such file or directory
/bin/sh: line 1: bc: command not found
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.15.5-arch1-1/build M=/var/lib/dkms/8188eu/5.3.9.r145.6146193/build  modules
make[1]: Entering directory '/usr/lib/modules/5.15.5-arch1-1/build'
cut: /etc/redhat-release: No such file or directory
/bin/sh: line 1: bc: command not found
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/mesh/rtw_mesh.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_rson.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_br_ext.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_sreset.o
  CC [M]  /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_btcoex_wifionly.o
/var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_br_ext.c:20:18: fatal error: net/ipx.h: No such file or directory
   20 |         #include <net/ipx.h>
      |                  ^~~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:277: /var/lib/dkms/8188eu/5.3.9.r145.6146193/build/core/rtw_br_ext.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:1868: /var/lib/dkms/8188eu/5.3.9.r145.6146193/build] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.15.5-arch1-1/build'
make: *** [Makefile:2062: modules] Error 2

_____________________________________________________________________________________________________
i dont know what to make of this

JoshuaJakowlew, 2021-12-01:
> ❯ sudo cat make.log DKMS make.log for 8188eu-5.3.9.r145.6146193 for kernel 5.15.5-arch1-1 (x86_64) Tue Nov 30 03:42:29 PM +0545 2021 cut: /etc/redhat-release: No such file or directory /bin/sh: line [...]

Looks similar to comment above. #898 doesn't help?

kimocoder, 2021-12-12:
Fixes for kernel v5.15-5+ has been merged. Update and build drivers again. If problem persists, open a new issue report. closed

closed 2021-12-12 by kimocoder