aircrack-ng/rtl8812au #917: Keep getting `DKMS tree already contains: 8812au-5.6.4.2_35491.20191025`

konsbn, 2021-11-28:
I am running Fedora 35 with the Linux 5.15.4 Kernel. For the 5.14.x kernels I was using this version of the [`rtl8812au`](https://github.com/KonstantIMP/rtl8812au). However it stopped working with the 5.15 kernel so I removed it using 'sudo make dkms_remove` and tried using the `aircrack-ng/rtl8812au`. Now when I try `sudo make dkms_install` I keep getting this error
```mkdir -p /usr/src/8812au-5.6.4.2_35491.20191025
cp -r * /usr/src/8812au-5.6.4.2_35491.20191025
dkms add -m 8812au -v 5.6.4.2_35491.20191025
Error! DKMS tree already contains: 8812au-5.6.4.2_35491.20191025
You cannot add the same module/version combo more than once.
make: *** [Makefile:2303: dkms_install] Error 3
```
I have also tried `make uninstall` as well as `sudo dkms remove 8812au/5.6.4.2_35491.20191025 --all` but the error still persists

konsbn, 2021-11-28:
Solution:
search your root directory for `88XX` and delete all the files manually! As well as the `88XX/` directory in `/var/lib/dkms`. Then you can run the following to delete all the other files
```
fd --type=f --exclude=tmp/ --exclude=/run/media/ 88XX | xargs sudo rm
```
then you can simply run the usual installation by changing to the source directory and following the readme instructions!

closed 2021-11-28 by konsbn

Sanaullah5, 2022-05-30:
samm to my problams plz help me my alfa modul 3001n 802.11n  usb realtek driver not working cant hwlop me

konsbn, 2022-08-28:
> samm to my problams plz help me my alfa modul 3001n 802.11n usb realtek driver not working cant hwlop me
Please try my  [solution](https://github.com/aircrack-ng/rtl8812au/issues/917#issuecomment-981101522) above

lakotamm, 2022-09-06:
> fd --type=f --exclude=tmp/ --exclude=/run/media/ 88XX | xargs sudo rm

this command failed to run on my Fedora 37. I tried to clean up dkms after I accidentally installed the module with headers for a different kernel version and failed the install.

However this worked:
`sudo rm -r /var/lib/dkms/8812au`

lakotamm, 2022-09-06:
> fd --type=f --exclude=tmp/ --exclude=/run/media/ 88XX | xargs sudo rm

this command failed to run on my Fedora 37. I tried to clean up dkms after I accidentally installed the module with headers for a different kernel version and failed the install.

However this worked:
sudo rm -r /var/lib/dkms/8812au

LilBamaBeets, 2022-12-20:
> > fd --type=f --exclude=tmp/ --exclude=/run/media/ 88XX | xargs sudo rm
> 
> this command failed to run on my Fedora 37. I tried to clean up dkms after I accidentally installed the module with headers for a different kernel version and failed the install.
> 
> However this worked: `sudo rm -r /var/lib/dkms/8812au`
> 
> It could be good to suggest running an update before installing dkms.

thank you you are amazing you can have my children

Nikolaikl, 2023-01-20:
For me it worked having removed it from the dkms tree, using make first and then make install

feiyuxiaoThu, 2025-12-05:
> thank you you are amazing you can have my children

I am pleased by this replay hhh.