aircrack-ng/rtl8812au #922: Errors on kernel version 5.15.5-76051505-generic

hanzy1110, 2021-12-01:
I've been trying to install the driver since the last update killed it a few days ago. 
I'm running Pop!OS 21.04 on a amd ryzen5 3600 cpu.
Here its the content of /var/log/syslog:
`ec  1 16:54:56 pop-os systemd[1]: Starting Network Manager...
Dec  1 16:54:56 pop-os gnome-shell[2079]: Object St.BoxLayout (0x5616e3cc3570), has been already deallocated — impossible to set any property on it. This might be caused by the object having been destroyed from C code using something such as destroy(), dispose(), or remove() vfuncs.
Dec  1 16:54:56 pop-os gnome-shell[2079]: Object St.BoxLayout (0x5616e3840f80), has been already deallocated — impossible to set any property on it. This might be caused by the object having been destroyed from C code using something such as destroy(), dispose(), or remove() vfuncs.
Dec  1 16:54:56 pop-os gnome-shell[2079]: Device state invalid, is 0
Dec  1 16:54:56 pop-os gnome-shell[2079]: Object St.Label (0x5616e329c1b0), has been already deallocated — impossible to set any property on it. This might be caused by the object having been destroyed from C code using something such as destroy(), dispose(), or remove() vfuncs.
Dec  1 16:54:56 pop-os gnome-shell[2079]: Object St.Icon (0x5616e24b4760), has been already deallocated — impossible to set any property on it. This might be caused by the object having been destroyed from C code using something such as destroy(), dispose(), or remove() vfuncs.
Dec  1 16:54:56 pop-os NetworkManager[26647]: <info>  [1638388496.9072] NetworkManager (version 1.30.0) is starting... (after a restart)
Dec  1 16:54:56 pop-os NetworkManager[26647]: <info>  [1638388496.9072] Read config: /etc/NetworkManager/NetworkManager.conf (lib: 10-dns-resolved.conf, 20-connectivity-pop.conf, no-mac-addr-change.conf) (etc: 10-globally-managed-devices.conf, default-wifi-powersave-on.conf)
Dec  1 16:54:56 pop-os NetworkManager[26647]: <info>  [1638388496.9082] bus-manager: acquired D-Bus service "org.freedesktop.NetworkManager"
Dec  1 16:54:56 pop-os systemd[1]: Started Network Manager.
Dec  1 16:54:56 pop-os NetworkManager[26647]: <info>  [1638388496.9105] manager[0x560fae960020]: monitoring kernel firmware directory '/lib/firmware'.
Dec  1 16:54:56 pop-os NetworkManager[26647]: <info>  [1638388496.9105] monitoring ifupdown state file '/run/network/ifstate'.
Dec  1 16:54:56 pop-os dbus-daemon[1102]: [system] Activating via systemd: service name='org.freedesktop.hostname1' unit='dbus-org.freedesktop.hostname1.service' requested by ':1.277' (uid=0 pid=26647 comm="/usr/sbin/NetworkManager --no-daemon ")
Dec  1 16:54:56 pop-os systemd[1]: Starting Hostname Service...
Dec  1 16:54:56 pop-os dbus-daemon[1102]: [system] Successfully activated service 'org.freedesktop.hostname1'
Dec  1 16:54:56 pop-os systemd[1]: Started Hostname Service.
Dec  1 16:54:56 pop-os NetworkManager[26647]: <info>  [1638388496.9720] hostname: hostname: using hostnamed
Dec  1 16:54:56 pop-os NetworkManager[26647]: <info>  [1638388496.9721] hostname: hostname changed from (none) to "pop-os"
Dec  1 16:54:56 pop-os NetworkManager[26647]: <info>  [1638388496.9722] dns-mgr[0x560fae94c170]: init: dns=systemd-resolved rc-manager=unmanaged (auto), plugin=systemd-resolved
Dec  1 16:54:56 pop-os NetworkManager[26647]: <info>  [1638388496.9726] rfkill9: found Wi-Fi radio killswitch (at /sys/devices/pci0000:00/0000:00:01.3/0000:03:00.0/usb1/1-2/1-2:1.0/ieee80211/phy9/rfkill9) (driver rtl88XXau)
Dec  1 16:54:56 pop-os NetworkManager[26647]: <info>  [1638388496.9727] manager[0x560fae960020]: rfkill: Wi-Fi hardware radio set enabled
Dec  1 16:54:56 pop-os NetworkManager[26647]: <info>  [1638388496.9727] manager[0x560fae960020]: rfkill: WWAN hardware radio set enabled
Dec  1 16:54:56 pop-os NetworkManager[26647]: <info>  [1638388496.9731] Loaded device plugin: NMAtmManager (/usr/lib/x86_64-linux-gnu/NetworkManager/1.30.0/libnm-device-plugin-adsl.so)
Dec  1 16:54:56 pop-os NetworkManager[26647]: <info>  [1638388496.9734] Loaded device plugin: NMTeamFactory (/usr/lib/x86_64-linux-gnu/NetworkManager/1.30.0/libnm-device-plugin-team.so)
Dec  1 16:54:56 pop-os NetworkManager[26647]: <info>  [1638388496.9738] Loaded device plugin: NMWwanFactory (/usr/lib/x86_64-linux-gnu/NetworkManager/1.30.0/libnm-device-plugin-wwan.so)
Dec  1 16:54:56 pop-os NetworkManager[26647]: <info>  [1638388496.9740] Loaded device plugin: NMWifiFactory (/usr/lib/x86_64-linux-gnu/NetworkManager/1.30.0/libnm-device-plugin-wifi.so)
Dec  1 16:54:56 pop-os NetworkManager[26647]: <info>  [1638388496.9742] Loaded device plugin: NMBluezManager (/usr/lib/x86_64-linux-gnu/NetworkManager/1.30.0/libnm-device-plugin-bluetooth.so)
Dec  1 16:54:56 pop-os NetworkManager[26647]: <info>  [1638388496.9743] manager: rfkill: Wi-Fi enabled by radio killswitch; enabled by state file
Dec  1 16:54:56 pop-os NetworkManager[26647]: <info>  [1638388496.9744] manager: rfkill: WWAN enabled by radio killswitch; enabled by state file
Dec  1 16:54:56 pop-os NetworkManager[26647]: <info>  [1638388496.9744] manager: Networking is enabled by state file
Dec  1 16:54:56 pop-os NetworkManager[26647]: <info>  [1638388496.9745] dhcp-init: Using DHCP client 'internal'
Dec  1 16:54:56 pop-os NetworkManager[26647]: <info>  [1638388496.9746] settings: Loaded settings plugin: ifupdown ("/usr/lib/x86_64-linux-gnu/NetworkManager/1.30.0/libnm-settings-plugin-ifupdown.so")
Dec  1 16:54:56 pop-os NetworkManager[26647]: <info>  [1638388496.9746] settings: Loaded settings plugin: keyfile (internal)
Dec  1 16:54:56 pop-os NetworkManager[26647]: <info>  [1638388496.9746] ifupdown: management mode: unmanaged
Dec  1 16:54:56 pop-os NetworkManager[26647]: <info>  [1638388496.9746] ifupdown:       interface-parser: parsing file /etc/network/interfaces
Dec  1 16:54:56 pop-os NetworkManager[26647]: <info>  [1638388496.9746] ifupdown:       interface-parser: source line includes interfaces file(s) /etc/network/interfaces.d
Dec  1 16:54:56 pop-os NetworkManager[26647]: <info>  [1638388496.9746] ifupdown:       interface-parser: finished parsing file /etc/network/interfaces
Dec  1 16:54:56 pop-os NetworkManager[26647]: <info>  [1638388496.9769] device (lo): carrier: link connected
Dec  1 16:54:56 pop-os NetworkManager[26647]: <info>  [1638388496.9770] manager: (lo): new Generic device (/org/freedesktop/NetworkManager/Devices/1)
Dec  1 16:54:56 pop-os NetworkManager[26647]: <info>  [1638388496.9775] manager: (enp37s0): new Ethernet device (/org/freedesktop/NetworkManager/Devices/2)
Dec  1 16:54:56 pop-os NetworkManager[26647]: <info>  [1638388496.9780] device (enp37s0): state change: unmanaged -> unavailable (reason 'managed', sys-iface-state: 'external')
Dec  1 16:54:56 pop-os NetworkManager[26647]: <info>  [1638388496.9789] device (usb0): carrier: link connected
Dec  1 16:54:56 pop-os NetworkManager[26647]: <info>  [1638388496.9791] manager: (usb0): new Ethernet device (/org/freedesktop/NetworkManager/Devices/3)
Dec  1 16:54:56 pop-os NetworkManager[26647]: <info>  [1638388496.9796] manager: (usb0): assume: will attempt to assume matching connection 'Wired connection 2' (6e7101f6-2b4f-3280-a05e-159554eb7fd0) (indicated)
Dec  1 16:54:56 pop-os NetworkManager[26647]: <info>  [1638388496.9797] device (usb0): state change: unmanaged -> unavailable (reason 'connection-assumed', sys-iface-state: 'assume')
Dec  1 16:54:56 pop-os NetworkManager[26647]: <info>  [1638388496.9801] device (usb0): state change: unavailable -> disconnected (reason 'connection-assumed', sys-iface-state: 'assume')
Dec  1 16:54:56 pop-os NetworkManager[26647]: <info>  [1638388496.9806] device (usb0): Activation: starting connection 'Wired connection 2' (6e7101f6-2b4f-3280-a05e-159554eb7fd0)
Dec  1 16:54:56 pop-os NetworkManager[26647]: <info>  [1638388496.9806] device (wlx00e04c6c48a1): driver supports Access Point (AP) mode
Dec  1 16:54:56 pop-os NetworkManager[26647]: <info>  [1638388496.9809] manager: (wlx00e04c6c48a1): new 802.11 Wi-Fi device (/org/freedesktop/NetworkManager/Devices/4)
Dec  1 16:54:56 pop-os NetworkManager[26647]: <info>  [1638388496.9813] device (wlx00e04c6c48a1): state change: unmanaged -> unavailable (reason 'managed', sys-iface-state: 'external')
Dec  1 16:54:57 pop-os NetworkManager[26647]: <warn>  [1638388497.5580] platform-linux: do-change-link[16]: failure changing link: failure 1 (Operation not permitted)
Dec  1 16:54:57 pop-os NetworkManager[26647]: <info>  [1638388497.5704] modem-manager: ModemManager available
[... cut ...]
Dec  1 16:55:53 pop-os wpa_supplicant[1152]: wlx00e04c6c48a1: Failed to initialize driver interface
Dec  1 16:55:53 pop-os NetworkManager[26647]: <error> [1638388553.8146] device (wlx00e04c6c48a1): Couldn't initialize supplicant interface: GDBus.Error:fi.w1.wpa_supplicant1.UnknownError: wpa_supplicant couldn't grab this interface.
Dec  1 16:55:53 pop-os NetworkManager[26647]: <info>  [1638388553.8146] device (wlx00e04c6c48a1): supplicant interface keeps failing, giving up`
There are simmilar issues but compilation on this kernel version seems to have been solved on the last commit

pablonunessilva, 2021-12-10:
same problem

kimocoder, 2021-12-12:
Just upload the dmesg log in dmesg.txt instead, I dont wanna read that one :/

kimocoder, 2021-12-12:
Pushed some kernel 5.15-5 fixes. git pull and rebuilt driver.

Open a new issue if the problem persists, but here it's now compiling and working without issues.

closed 2021-12-12 by kimocoder