aircrack-ng/rtl8812au #926: Compile fails on Nvidia Jetson board.

Jcwscience, 2021-12-18:
When I try to compile the driver I get these warnings.

```

Building modules, stage 2.
  MODPOST 1 modules
WARNING: "cfg80211_mgmt_tx_status" [/home/john/HackCopter/rtl8812au/88XXau.ko] undefined!
WARNING: "wiphy_apply_custom_regulatory" [/home/john/HackCopter/rtl8812au/88XXau.ko] undefined!
WARNING: "ieee80211_frequency_to_channel" [/home/john/HackCopter/rtl8812au/88XXau.ko] undefined!
WARNING: "cfg80211_rx_mgmt" [/home/john/HackCopter/rtl8812au/88XXau.ko] undefined!
WARNING: "cfg80211_new_sta" [/home/john/HackCopter/rtl8812au/88XXau.ko] undefined!
WARNING: "cfg80211_inform_bss_frame_data" [/home/john/HackCopter/rtl8812au/88XXau.ko] undefined!
WARNING: "ieee80211_radiotap_iterator_next" [/home/john/HackCopter/rtl8812au/88XXau.ko] undefined!
WARNING: "cfg80211_unlink_bss" [/home/john/HackCopter/rtl8812au/88XXau.ko] undefined!
WARNING: "cfg80211_put_bss" [/home/john/HackCopter/rtl8812au/88XXau.ko] undefined!
WARNING: "cfg80211_roamed" [/home/john/HackCopter/rtl8812au/88XXau.ko] undefined!
WARNING: "cfg80211_scan_done" [/home/john/HackCopter/rtl8812au/88XXau.ko] undefined!
WARNING: "ieee80211_radiotap_iterator_init" [/home/john/HackCopter/rtl8812au/88XXau.ko] undefined!
WARNING: "cfg80211_ibss_joined" [/home/john/HackCopter/rtl8812au/88XXau.ko] undefined!
WARNING: "cfg80211_michael_mic_failure" [/home/john/HackCopter/rtl8812au/88XXau.ko] undefined!
WARNING: "cfg80211_disconnected" [/home/john/HackCopter/rtl8812au/88XXau.ko] undefined!
WARNING: "cfg80211_get_bss" [/home/john/HackCopter/rtl8812au/88XXau.ko] undefined!
WARNING: "wiphy_new_nm" [/home/john/HackCopter/rtl8812au/88XXau.ko] undefined!
WARNING: "wiphy_free" [/home/john/HackCopter/rtl8812au/88XXau.ko] undefined!
WARNING: "__ieee80211_get_channel" [/home/john/HackCopter/rtl8812au/88XXau.ko] undefined!
WARNING: "cfg80211_ready_on_channel" [/home/john/HackCopter/rtl8812au/88XXau.ko] undefined!
WARNING: "cfg80211_ch_switch_started_notify" [/home/john/HackCopter/rtl8812au/88XXau.ko] undefined!
WARNING: "cfg80211_ch_switch_notify" [/home/john/HackCopter/rtl8812au/88XXau.ko] undefined!
WARNING: "cfg80211_connect_bss" [/home/john/HackCopter/rtl8812au/88XXau.ko] undefined!
WARNING: "wiphy_unregister" [/home/john/HackCopter/rtl8812au/88XXau.ko] undefined!
WARNING: "cfg80211_del_sta_sinfo" [/home/john/HackCopter/rtl8812au/88XXau.ko] undefined!
WARNING: "cfg80211_remain_on_channel_expired" [/home/john/HackCopter/rtl8812au/88XXau.ko] undefined!
WARNING: "ieee80211_channel_to_frequency" [/home/john/HackCopter/rtl8812au/88XXau.ko] undefined!
WARNING: "wiphy_register" [/home/john/HackCopter/rtl8812au/88XXau.ko] undefined!
  CC      /home/john/HackCopter/rtl8812au/88XXau.mod.o
  LD [M]  /home/john/HackCopter/rtl8812au/88XXau.ko
make[1]: Leaving directory '/home/john/Linux_for_Tegra/source/public/kernel/kernel-4.9'

```

Is there anything I can do to fix this?
Thanks

closed 2021-12-19 by Jcwscience

lsy512, 2023-02-20:
@Jcwscience hi, i meet the same problem,how did you solved th problem?

Jcwscience, 2023-02-21:
You know I’m not sure. I’ll try to recreate my steps in a few days and see if I can find out what I did

lsy512, 2023-02-21:
Look forward to your good news