aircrack-ng/rtl8812au #928: Your kernel headers for kernel 5.10.43v64 cannot be found at /lib/modules/5.10.43v64/build or /lib/modules/5.10.43v64/source.

cray12399, 2021-12-20:
I am trying to install this driver on a Raspberry Pi 4 with the latest version of Raspbian using BerryBoot. However, I am getting the following output:

`mkdir -p /usr/src/8812au-5.6.4.2_35491.20191025
cp -r * /usr/src/8812au-5.6.4.2_35491.20191025
dkms add -m 8812au -v 5.6.4.2_35491.20191025

Creating symlink /var/lib/dkms/8812au/5.6.4.2_35491.20191025/source ->
                 /usr/src/8812au-5.6.4.2_35491.20191025

DKMS: add completed.
dkms build -m 8812au -v 5.6.4.2_35491.20191025
Error! echo
Your kernel headers for kernel 5.10.43v64 cannot be found at
/lib/modules/5.10.43v64/build or /lib/modules/5.10.43v64/source.
You can use the --kernelsourcedir option to tell DKMS where it's located.
make: *** [Makefile:2304: dkms_install] Error 1`

I followed the steps exactly as instructed on the README. So, I have no idea what is making this not work.

XZCendence, 2022-06-17:
Did you ever get this to work?

XZCendence, 2022-06-17:
I'm also on a pi 4 using berryboot and running into the same issue.