aircrack-ng/rtl8812au #933: Issue with raspberry pi beta kernel

Jcwscience, 2022-01-10:
I am attempting to build this driver on a beta version of the raspberry pi kernel, and I’m getting these errors.

```
ERROR: modpost: "__const_udelay" [/home/pi/rtl8812au/88XXau.ko] undefined!
ERROR: modpost: "__arch_copy_from_user" [/home/pi/rtl8812au/88XXau.ko] undefined!
ERROR: modpost: "_raw_spin_unlock" [/home/pi/rtl8812au/88XXau.ko] undefined!
ERROR: modpost: "__rcu_read_lock" [/home/pi/rtl8812au/88XXau.ko] undefined!
ERROR: modpost: "__rcu_read_unlock" [/home/pi/rtl8812au/88XXau.ko] undefined!
ERROR: modpost: "cpu_hwcap_keys" [/home/pi/rtl8812au/88XXau.ko] undefined!
ERROR: modpost: "csum_ipv6_magic" [/home/pi/rtl8812au/88XXau.ko] undefined!
ERROR: modpost: "__warn_printk" [/home/pi/rtl8812au/88XXau.ko] undefined!
ERROR: modpost: "arm64_const_caps_ready" [/home/pi/rtl8812au/88XXau.ko] undefined!
ERROR: modpost: "__udelay" [/home/pi/rtl8812au/88XXau.ko] undefined!
ERROR: modpost: "preempt_schedule_notrace" [/home/pi/rtl8812au/88XXau.ko] undefined!
ERROR: modpost: "__arch_copy_to_user" [/home/pi/rtl8812au/88XXau.ko] undefined!
```

Could anyone point me in the right direction to get this working?

DoM1niC, 2022-02-16:
same issue here

cantudo, 2022-04-21:
I had this same issue while trying to compile an RTC driver module. I used rpi-source to get the kernel sources, it turns out that rpi-source as of now only works for 32 bit architecture. 

This [pull request](https://github.com/RPi-Distro/rpi-source/pull/2) addresses this and adds support for 64 bit, arch64, you can download the upgraded rpi-source from [here](https://raw.githubusercontent.com/RPi-Distro/rpi-source/ceee03b5fb9cee65dd933f4784bb455cfa872a76/rpi-source). You can then execute it with ```python2 rpi-source --architecture 1```, indicating that you wanna build for a 64 bit architecture, it will download the kernel source and generate a ```Modules.symvers``` file, which will contain the functions you are missing, (__arch_copy_from_user, __arch_copy_from_user, etc.). You can then try to build this driver. I hope this helps you.

Zahrun, 2023-10-17:
Thanks @cantudo. This still works in 2023 with kernel 6.1.57-v8+. Quite sad that the PR wasn’t merged.