aircrack-ng/rtl8812au #943: Needing Help.

S1m1B1lly, 2022-02-16:
**How do i compile drivers and install them on android? there is no guide anywhere.**

closed 2024-04-09 by S1m1B1lly

kimocoder, 2024-04-09:
Like I did here;

https://github.com/kimocoder/kernel_common/commit/859c386cc2d7dcfb30452949b92631f98ff35aa2

Just take it as a patch like this (in root Android kernel folder);
wget https://github.com/kimocoder/kernel_common/commit/859c386cc2d7dcfb30452949b92631f98ff35aa2.patch && patch -p1 < *. patch

vaibhav423, 2025-10-31:
@kimocoder the page isnt available .
i tried to compile but after loading .ko it didnt work

[10150.086607] 88XXau: no symbol version for module_layout
[10150.086637] 88XXau: loading out-of-tree module taints kernel.