aircrack-ng/rtl8812au #950: Build on raspberry pi 4 fails with fatal error: asm/types.h: No such file or directory

lukaszwielgosz, 2022-03-26:
Building:
```
pi@raspberrypi:~/rtl8812au $ sudo make dkms_install
mkdir: created directory '/usr/src/8812au-5.6.4.2_35491.20191025'
cp -r * /usr/src/8812au-5.6.4.2_35491.20191025
dkms add -m 8812au -v 5.6.4.2_35491.20191025

Creating symlink /var/lib/dkms/8812au/5.6.4.2_35491.20191025/source ->
                 /usr/src/8812au-5.6.4.2_35491.20191025

DKMS: add completed.
dkms build -m 8812au -v 5.6.4.2_35491.20191025

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' -j4 KVER=5.10.103-v7l+ KSRC=/lib/modules/5.10.103-v7l+/build....(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.10.103-v7l+ (armv7l)
Consult /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/make.log for more information.
make: *** [Makefile:2304: dkms_install] Error 10
```

make.log contents:
```
pi@raspberrypi:~/rtl8812au $ cat /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/make.log
DKMS make.log for 8812au-5.6.4.2_35491.20191025 for kernel 5.10.103-v7l+ (armv7l)
Sat Mar 26 07:26:04 GMT 2022
make[1]: Entering directory '/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build'
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/5.10.103-v7l+/build M=/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build  modules
make[2]: Entering directory '/usr/src/linux-headers-5.10.103-v7l+'
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_io.o
In file included from ./include/linux/types.h:6,
                 from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/include/basic_types.h:40,
                 from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/include/drv_types.h:25,
                 from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_cmd.c:17:
./include/uapi/linux/types.h:5:10: fatal error: asm/types.h: No such file or directory
    5 | #include <asm/types.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
In file included from ./include/linux/types.h:6,
                 from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/include/basic_types.h:40,
                 from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/include/drv_types.h:25,
                 from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_security.c:17:
./include/uapi/linux/types.h:5:10: fatal error: asm/types.h: No such file or directory
    5 | #include <asm/types.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
make[3]: *** [scripts/Makefile.build:280: /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_cmd.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[3]: *** [scripts/Makefile.build:280: /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_security.o] Error 1
In file included from ./include/linux/types.h:6,
                 from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/include/basic_types.h:40,
                 from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/include/drv_types.h:25,
                 from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_debug.c:17:
./include/uapi/linux/types.h:5:10: fatal error: asm/types.h: No such file or directory
    5 | #include <asm/types.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
make[3]: *** [scripts/Makefile.build:280: /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_debug.o] Error 1
In file included from ./include/linux/types.h:6,
                 from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/include/basic_types.h:40,
                 from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/include/drv_types.h:25,
                 from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_io.c:47:
./include/uapi/linux/types.h:5:10: fatal error: asm/types.h: No such file or directory
    5 | #include <asm/types.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
make[3]: *** [scripts/Makefile.build:280: /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_io.o] Error 1
make[2]: *** [Makefile:1825: /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-5.10.103-v7l+'
make[1]: *** [Makefile:2244: modules] Error 2
make[1]: Leaving directory '/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build'

```

lukaszwielgosz, 2022-03-26:
So the problem appears only when you set in Makefile:
```
CONFIG_PLATFORM_ARM_RPI = n
CONFIG_PLATFORM_ARM64_RPI = y
```

but when you set:
```
CONFIG_PLATFORM_ARM_RPI = y
CONFIG_PLATFORM_ARM64_RPI = n
```
Then it compiles and works fine.

alexkutsan, 2022-04-16:
I have mostely similar issue. 
BUt trich with 
```
CONFIG_PLATFORM_ARM_RPI = y
CONFIG_PLATFORM_ARM64_RPI = n
```

Does not help for me. 

In my case compiler is not able to find anotehr file in asm dir 
```
In file included from ./include/linux/spinlock.h:50:0,
                 from ./include/linux/seqlock.h:35,
                 from ./include/linux/time.h:5,
                 from ./include/linux/stat.h:18,
                 from ./include/linux/module.h:10,
                 from /home/pi/rtl8812au/include/basic_types.h:41,
                 from /home/pi/rtl8812au/include/drv_types.h:25,
                 from /home/pi/rtl8812au/core/rtw_cmd.c:17:
./include/linux/preempt.h:59:25: fatal error: asm/preempt.h: No such file or directory
```
Full log : 
```make ARCH=arm CROSS_COMPILE= -C /lib/modules/4.9.119-BPI-M4-Kernel/build M=/home/pi/rtl8812au  modules
make[1]: Entering directory '/usr/src/linux-headers-4.9.119-BPI-M4-Kernel'
  CC [M]  /home/pi/rtl8812au/core/rtw_cmd.o
In file included from ./include/linux/irqflags.h:15:0,
                 from ./arch/arm/include/asm/bitops.h:27,
                 from ./include/linux/bitops.h:36,
                 from ./include/linux/kernel.h:10,
                 from ./include/linux/list.h:8,
                 from ./include/linux/module.h:9,
                 from /home/pi/rtl8812au/include/basic_types.h:41,
                 from /home/pi/rtl8812au/include/drv_types.h:25,
                 from /home/pi/rtl8812au/core/rtw_cmd.c:17:
./arch/arm/include/asm/irqflags.h:21:5: warning: "__LINUX_ARM_ARCH__" is not defined [-Wundef]
 #if __LINUX_ARM_ARCH__ >= 6
     ^~~~~~~~~~~~~~~~~~
In file included from ./arch/arm/include/asm/bitops.h:28:0,
                 from ./include/linux/bitops.h:36,
                 from ./include/linux/kernel.h:10,
                 from ./include/linux/list.h:8,
                 from ./include/linux/module.h:9,
                 from /home/pi/rtl8812au/include/basic_types.h:41,
                 from /home/pi/rtl8812au/include/drv_types.h:25,
                 from /home/pi/rtl8812au/core/rtw_cmd.c:17:
./arch/arm/include/asm/barrier.h:12:5: warning: "__LINUX_ARM_ARCH__" is not defined [-Wundef]
 #if __LINUX_ARM_ARCH__ >= 7 ||  \
     ^~~~~~~~~~~~~~~~~~
./arch/arm/include/asm/barrier.h:13:3: warning: "__LINUX_ARM_ARCH__" is not defined [-Wundef]
  (__LINUX_ARM_ARCH__ == 6 && defined(CONFIG_CPU_32v6K))
   ^~~~~~~~~~~~~~~~~~
./arch/arm/include/asm/barrier.h:19:5: warning: "__LINUX_ARM_ARCH__" is not defined [-Wundef]
 #if __LINUX_ARM_ARCH__ >= 7
     ^~~~~~~~~~~~~~~~~~
./arch/arm/include/asm/barrier.h:23:35: warning: "__LINUX_ARM_ARCH__" is not defined [-Wundef]
 #elif defined(CONFIG_CPU_XSC3) || __LINUX_ARM_ARCH__ == 6
                                   ^~~~~~~~~~~~~~~~~~
In file included from ./include/linux/bitops.h:36:0,
                 from ./include/linux/kernel.h:10,
                 from ./include/linux/list.h:8,
                 from ./include/linux/module.h:9,
                 from /home/pi/rtl8812au/include/basic_types.h:41,
                 from /home/pi/rtl8812au/include/drv_types.h:25,
                 from /home/pi/rtl8812au/core/rtw_cmd.c:17:
./arch/arm/include/asm/bitops.h:215:5: warning: "__LINUX_ARM_ARCH__" is not defined [-Wundef]
 #if __LINUX_ARM_ARCH__ < 5
     ^~~~~~~~~~~~~~~~~~
In file included from ./arch/arm/include/asm/bitops.h:218:0,
                 from ./include/linux/bitops.h:36,
                 from ./include/linux/kernel.h:10,
[... 70 lines cut ...]
                 from /home/pi/rtl8812au/include/drv_types.h:25,
                 from /home/pi/rtl8812au/core/rtw_cmd.c:17:
./arch/arm/include/asm/swab.h:20:5: warning: "__LINUX_ARM_ARCH__" is not defined [-Wundef]
 #if __LINUX_ARM_ARCH__ >= 6
     ^~~~~~~~~~~~~~~~~~
In file included from ./include/linux/spinlock.h:50:0,
                 from ./include/linux/seqlock.h:35,
                 from ./include/linux/time.h:5,
                 from ./include/linux/stat.h:18,
                 from ./include/linux/module.h:10,
                 from /home/pi/rtl8812au/include/basic_types.h:41,
                 from /home/pi/rtl8812au/include/drv_types.h:25,
                 from /home/pi/rtl8812au/core/rtw_cmd.c:17:
./include/linux/preempt.h:59:25: fatal error: asm/preempt.h: No such file or directory
 #include <asm/preempt.h>
                         ^
compilation terminated.
scripts/Makefile.build:334: recipe for target '/home/pi/rtl8812au/core/rtw_cmd.o' failed
make[2]: *** [/home/pi/rtl8812au/core/rtw_cmd.o] Error 1
Makefile:1613: recipe for target '_module_/home/pi/rtl8812au' failed
make[1]: *** [_module_/home/pi/rtl8812au] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.9.119-BPI-M4-Kernel'
Makefile:2244: recipe for target 'modules' failed
make: *** [modules] Error 2
```

microinteractive, 2022-09-13:
This config:
```
CONFIG_PLATFORM_ARM_RPI = y
CONFIG_PLATFORM_ARM64_RPI = n
```
I'm getting this error:

``` console
pi@malina:~/rtl8812au $ sudo make dkms_install
mkdir: created directory '/usr/src/8812au-5.6.4.2_35491.20191025'
cp -r * /usr/src/8812au-5.6.4.2_35491.20191025
dkms add -m 8812au -v 5.6.4.2_35491.20191025

Creating symlink /var/lib/dkms/8812au/5.6.4.2_35491.20191025/source ->
                 /usr/src/8812au-5.6.4.2_35491.20191025

DKMS: add completed.
dkms build -m 8812au -v 5.6.4.2_35491.20191025

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' -j4 KVER=5.15.61-v7+ KSRC=/lib/modules/5.15.61-v7+/build...........(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.15.61-v7+ (armv7l)
Consult /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/make.log for more information.
make: *** [Makefile:2317: dkms_install] Error 10
```
Full log: /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/make.log
``` log
DKMS make.log for 8812au-5.6.4.2_35491.20191025 for kernel 5.15.61-v7+ (armv7l)
Wed 14 Sep 01:19:18 CEST 2022
make[1]: Entering directory '/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build'
make ARCH=arm CROSS_COMPILE= -C /lib/modules/5.15.61-v7+/build M=/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build  modules
make[2]: Entering directory '/usr/src/linux-headers-5.15.61-v7+'
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_io.o
In file included from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/include/drv_types.h:73,
                 from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_debug.c:17:
/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/include/rtw_recv.h: In function ‘recvframe_put’:
/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/include/rtw_recv.h:750:17: warning: variable ‘prev_rx_tail’ set but not used [-Wunused-but-set-variable]
  750 |  unsigned char *prev_rx_tail;
      |                 ^~~~~~~~~~~~
In file included from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/include/drv_types.h:73,
                 from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_io.c:47:
/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/include/rtw_recv.h: In function ‘recvframe_put’:
/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/include/rtw_recv.h:750:17: warning: variable ‘prev_rx_tail’ set but not used [-Wunused-but-set-variable]
  750 |  unsigned char *prev_rx_tail;
      |                 ^~~~~~~~~~~~
In file included from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/include/drv_types.h:73,
                 from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_security.c:17:
/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/include/rtw_recv.h: In function ‘recvframe_put’:
/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/include/rtw_recv.h:750:17: warning: variable ‘prev_rx_tail’ set but not used [-Wunused-but-set-variable]
  750 |  unsigned char *prev_rx_tail;
      |                 ^~~~~~~~~~~~
In file included from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/include/drv_types.h:73,
                 from /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_cmd.c:17:
/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/include/rtw_recv.h: In function ‘recvframe_put’:
/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/include/rtw_recv.h:750:17: warning: variable ‘prev_rx_tail’ set but not used [-Wunused-but-set-variable]
  750 |  unsigned char *prev_rx_tail;
      |                 ^~~~~~~~~~~~
/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_security.c: In function ‘rtw_tkip_encrypt’:
/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_security.c:684:6: warning: variable ‘prwskeylen’ set but not used [-Wunused-but-set-variable]
  684 |  u32 prwskeylen;
      |      ^~~~~~~~~~
/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_security.c: In function ‘rtw_tkip_decrypt’:
/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_security.c:802:6: warning: variable ‘prwskeylen’ set but not used [-Wunused-but-set-variable]
  802 |  u32 prwskeylen;
[... 20 lines cut ...]
      |                                       ^~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_cmd.c:1484:32: warning: taking address of packed member of ‘struct _WLAN_BSSID_EX’ may result in an unaligned pointer value [-Waddress-of-packed-member]
 1484 |    pnetwork->network.IELength, &psecnetwork->IELength);
      |                                ^~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_cmd.c:1489:57: warning: taking address of packed member of ‘struct _WLAN_BSSID_EX’ may result in an unaligned pointer value [-Waddress-of-packed-member]
 1489 |  rtw_append_exented_cap(padapter, &psecnetwork->IEs[0], &psecnetwork->IELength);
      |                                                         ^~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_cmd.c:1331:6: warning: variable ‘auth’ set but not used [-Wunused-but-set-variable]
 1331 |  u8 *auth, res = _SUCCESS;
      |      ^~~~
as: error while loading shared libraries: /lib/arm-linux-gnueabihf/libbfd-2.35.2-system.so: invalid ELF header
[... cut ...]
as: error while loading shared libraries: /lib/arm-linux-gnueabihf/libbfd-2.35.2-system.so: invalid ELF header
make[3]: *** [scripts/Makefile.build:289: /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_debug.o] Error 1
make[2]: *** [Makefile:1898: /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-5.15.61-v7+'
make[1]: *** [Makefile:2257: modules] Error 2
make[1]: Leaving directory '/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build'
```
Is anyone has a solution for that?