aircrack-ng/rtl8812au #958: problem in monitor mode

lsy512, 2022-04-24:
hi,i am using this driver on ubuntu. after make install ,i do these steps to let rtl8812 in monitor mode

sudo ifconfig wlx74da381f48f7 down
sudo iwconfig wlx74da381f48f7 mode monitor
sudo ifconfig wlx74da381f48f7 up
sudo iwconfig wlx74da381f48f7 channel 4

then,i use tcpdump to moniter data
sudo tcpdump -i wlx74da381f48f7
i fount data is from different channel

[image]
[image]

i have set channel to 4.why i can receive data from other channal ?
can you help me to solve this problem?
thanks

lsy512, 2022-04-27:
Thanks for your help, it worked fine after I closed the wpa_supplicant program

ZerBea, 2022-04-27:
You're welcome. tcpdump is not able to notice that another tool change the channel. That caused that you received traffic from other channels set by the other tool.
Please close this issue, because it isn't related to the driver.

lsy512, 2022-04-28:
I tested again today and found a very strange problem.
I use a rtl8812 network card for data injection. Another RTL8812AU for data packet capture.
My test method is as follows：
1 inject data at greater than 5mbps on the transmitter side
2 Use the command tcpdump -i wlan -c 2000 to receive data to verify the data reception rate
In this case, I test that it takes about 15-20s to receive 2000 packets, which is obviously inconsistent with the injection rate.

Without finding the source of the problem, I guessed that it was a receiving problem, so I used another rt3070 to receive on the same device. It takes about 4-5s to receive 2000 packets. According to calculations this is about 5mbps.
At this time, I switched to the original rtl8812au for data reception on this device, and found that it only took 4-5s for the rtl8812 network card to receive 2000 data packets.

After repeated tests, I found that after removing the rt3070 network card, it takes 15-20s for the rtl8812 network card to receive 2000 packets, but after inserting the RT3070 and setting it to monitor mode. Then use rtl8812au for data reception, it only takes 4-5s at this time.
This is a strange problem, is it a system problem or a driver problem? 
thanks

closed 2022-05-24 by lsy512