aircrack-ng/rtl8812au #960: Problems with TP-LINK AC600 Archer T2U Nano

Aleadar, 2022-05-03:
Hi,

I'm trying to use this driver on my wifi adapter (see title). 

I can install and use normally, but a few moments he stop automatically and I need to reboot machine. Always after 3-5 minutes using he stop.

Someone know some solution?

zeph, 2022-05-10:
no need to reboot, but I'm also having a glitchy behaviour... lots of packet loss

maybe some more details on the distro and kernel u are running on?

morrownr, 2022-05-16:
Hi @zeph 

There is some confusion about what driver should be used in this case. The driver in this repo supports both the 8812au and 8821au/8811au chipsets. This is how it used to be in the older drivers that Realtek provided. A few years ago Realtek changed and now provides separate drivers. If using a site that uses the separate drivers, you have to make sure you go to the correct repo. Example:

AC600 - 8821au/8811au chipsets - https://github.com/morrownr/8821au-20210708
AC1200 - 8812au chipset - https://github.com/morrownr/8812au-20210629

Good luck