aircrack-ng/rtl8812au #964: build failed on kernel 5.18

amazingfate, 2022-05-26:
armbian/build#3814

Leont, 2022-05-30:
I'm observing the same problem. Bunch of errors along these lines:
~~~
ld: /var/lib/dkms/rtl88xxau/r1181.37e27f9/build/hal/phydm/rtl8814a/halhwimg8814a_rf.o: in function `is_multicast_mac_addr': 
/var/lib/dkms/rtl88xxau/r1181.37e27f9/build/include/ieee80211.h:1620: multiple definition of `is_multicast_mac_addr'; /var/lib/dkms/rtl88xxau/r1181.37e27f9/build/core/rtw_cmd.o:/var/lib/dkms/rtl88xxau/r1181.37e27f9/build/include/ieee80211.h:1620: first defined here 
ld: /var/lib/dkms/rtl88xxau/r1181.37e27f9/build/hal/phydm/rtl8814a/halhwimg8814a_rf.o: in function `is_broadcast_mac_addr': 
/var/lib/dkms/rtl88xxau/r1181.37e27f9/build/include/ieee80211.h:1625: multiple definition of `is_broadcast_mac_addr'; /var/lib/dkms/rtl88xxau/r1181.37e27f9/build/core/rtw_cmd.o:/var/lib/dkms/rtl88xxau/r1181.37e27f9/build/include/ieee80211.h:1625: first defined here 
ld: /var/lib/dkms/rtl88xxau/r1181.37e27f9/build/hal/phydm/rtl8814a/halhwimg8814a_rf.o: in function `is_zero_mac_addr': 
/var/lib/dkms/rtl88xxau/r1181.37e27f9/build/include/ieee80211.h:1631: multiple definition of `is_zero_mac_addr'; /var/lib/dkms/rtl88xxau/r1181.37e27f9/build/core/rtw_cmd.o:/var/lib/dkms/rtl88xxau/r1181.37e27f9/build/include/ieee80211.h:1631: first defined here 
ld: /var/lib/dkms/rtl88xxau/r1181.37e27f9/build/hal/phydm/halrf/rtl8814a/halrf_iqk_8814a.o: in function `is_multicast_mac_addr': 
/var/lib/dkms/rtl88xxau/r1181.37e27f9/build/include/ieee80211.h:1620: multiple definition of `is_multicast_mac_addr'; /var/lib/dkms/rtl88xxau/r1181.37e27f9/build/core/rtw_cmd.o:/var/lib/dkms/rtl88xxau/r1181.37e27f9/build/include/ieee80211.h:1620: first defined here 
ld: /var/lib/dkms/rtl88xxau/r1181.37e27f9/build/hal/phydm/halrf/rtl8814a/halrf_iqk_8814a.o: in function `is_broadcast_mac_addr': 
/var/lib/dkms/rtl88xxau/r1181.37e27f9/build/include/ieee80211.h:1625: multiple definition of `is_broadcast_mac_addr'; /var/lib/dkms/rtl88xxau/r1181.37e27f9/build/core/rtw_cmd.o:/var/lib/dkms/rtl88xxau/r1181.37e27f9/build/include/ieee80211.h:1625: first defined here 
ld: /var/lib/dkms/rtl88xxau/r1181.37e27f9/build/hal/phydm/halrf/rtl8814a/halrf_iqk_8814a.o: in function `is_zero_mac_addr': 
/var/lib/dkms/rtl88xxau/r1181.37e27f9/build/include/ieee80211.h:1631: multiple definition of `is_zero_mac_addr'; /var/lib/dkms/rtl88xxau/r1181.37e27f9/build/core/rtw_cmd.o:/var/lib/dkms/rtl88xxau/r1181.37e27f9/build/include/ieee80211.h:1631: first defined here 
ld: /var/lib/dkms/rtl88xxau/r1181.37e27f9/build/hal/phydm/rtl8814a/phydm_regconfig8814a.o: in function `is_multicast_mac_addr': 
/var/lib/dkms/rtl88xxau/r1181.37e27f9/build/include/ieee80211.h:1620: multiple definition of `is_multicast_mac_addr'; /var/lib/dkms/rtl88xxau/r1181.37e27f9/build/core/rtw_cmd.o:/var/lib/dkms/rtl88xxau/r1181.37e27f9/build/include/ieee80211.h:1620: first defined here 
ld: /var/lib/dkms/rtl88xxau/r1181.37e27f9/build/hal/phydm/rtl8814a/phydm_regconfig8814a.o: in function `is_broadcast_mac_addr': 
/var/lib/dkms/rtl88xxau/r1181.37e27f9/build/include/ieee80211.h:1625: multiple definition of `is_broadcast_mac_addr'; /var/lib/dkms/rtl88xxau/r1181.37e27f9/build/core/rtw_cmd.o:/var/lib/dkms/rtl88xxau/r1181.37e27f9/build/include/ieee80211.h:1625: first defined here 
ld: /var/lib/dkms/rtl88xxau/r1181.37e27f9/build/hal/phydm/rtl8814a/phydm_regconfig8814a.o: in function `is_zero_mac_addr': 
/var/lib/dkms/rtl88xxau/r1181.37e27f9/build/include/ieee80211.h:1631: multiple definition of `is_zero_mac_addr'; /var/lib/dkms/rtl88xxau/r1181.37e27f9/build/core/rtw_cmd.o:/var/lib/dkms/rtl88xxau/r1181.37e27f9/build/include/ieee80211.h:1631: first defined here 
ld: /var/lib/dkms/rtl88xxau/r1181.37e27f9/build/hal/phydm/halrf/rtl8814a/halrf_8814a_ce.o: in function `is_multicast_mac_addr': 
/var/lib/dkms/rtl88xxau/r1181.37e27f9/build/include/ieee80211.h:1620: multiple definition of `is_multicast_mac_addr'; /var/lib/dkms/rtl88xxau/r1181.37e27f9/build/core/rtw_cmd.o:/var/lib/dkms/rtl88xxau/r1181.37e27f9/build/include/ieee80211.h:1620: first defined here 
ld: /var/lib/dkms/rtl88xxau/r1181.37e27f9/build/hal/phydm/halrf/rtl8814a/halrf_8814a_ce.o: in function `is_broadcast_mac_addr': 
/var/lib/dkms/rtl88xxau/r1181.37e27f9/build/include/ieee80211.h:1625: multiple definition of `is_broadcast_mac_addr'; /var/lib/dkms/rtl88xxau/r1181.37e27f9/build/core/rtw_cmd.o:/var/lib/dkms/rtl88xxau/r1181.37e27f9/build/include/ieee80211.h:1625: first defined here 
ld: /var/lib/dkms/rtl88xxau/r1181.37e27f9/build/hal/phydm/halrf/rtl8814a/halrf_8814a_ce.o: in function `is_zero_mac_addr': 
/var/lib/dkms/rtl88xxau/r1181.37e27f9/build/include/ieee80211.h:1631: multiple definition of `is_zero_mac_addr'; /var/lib/dkms/rtl88xxau/r1181.37e27f9/build/core/rtw_cmd.o:/var/lib/dkms/rtl88xxau/r1181.37e27f9/build/include/ieee80211.h:1631: first defined here 
ld: /var/lib/dkms/rtl88xxau/r1181.37e27f9/build/hal/phydm/rtl8814a/phydm_rtl8814a.o: in function `is_multicast_mac_addr': 
/var/lib/dkms/rtl88xxau/r1181.37e27f9/build/include/ieee80211.h:1620: multiple definition of `is_multicast_mac_addr'; /var/lib/dkms/rtl88xxau/r1181.37e27f9/build/core/rtw_cmd.o:/var/lib/dkms/rtl88xxau/r1181.37e27f9/build/include/ieee80211.h:1620: first defined here 
ld: /var/lib/dkms/rtl88xxau/r1181.37e27f9/build/hal/phydm/rtl8814a/phydm_rtl8814a.o: in function `is_broadcast_mac_addr': 
/var/lib/dkms/rtl88xxau/r1181.37e27f9/build/include/ieee80211.h:1625: multiple definition of `is_broadcast_mac_addr'; /var/lib/dkms/rtl88xxau/r1181.37e27f9/build/core/rtw_cmd.o:/var/lib/dkms/rtl88xxau/r1181.37e27f [... cut ...]
/var/lib/dkms/rtl88xxau/r1181.37e27f9/build/include/ieee80211.h:1620: multiple definition of `is_multicast_mac_addr'; /var/lib/dkms/rtl88xxau/r1181.37e27f9/build/core/rtw_cmd.o:/var/lib/dkms/rtl88xxau/r1181.37e27f9/build/include/ieee80211.h:1620: first defined here 
ld: /var/lib/dkms/rtl88xxau/r1181.37e27f9/build/hal/phydm/txbf/haltxbf8814a.o: in function `is_broadcast_mac_addr': 
/var/lib/dkms/rtl88xxau/r1181.37e27f9/build/include/ieee80211.h:1625: multiple definition of `is_broadcast_mac_addr'; /var/lib/dkms/rtl88xxau/r1181.37e27f9/build/core/rtw_cmd.o:/var/lib/dkms/rtl88xxau/r1181.37e27f9/build/include/ieee80211.h:1625: first defined here 
~~~

CGarces, 2022-05-30:
Please apply PR #961 and #959 to fix the compilation errors.

folfgirl, 2022-06-03:
I am having the same issue

kimocoder, 2022-06-06:
this issue should be resolved. please update.

closed 2022-06-06 by kimocoder