aircrack-ng/rtl8812au #967: skbuff: skb_over_panic error

lsy512, 2022-06-02:
hi
i am using  rtl8812au.after i set it to monitor mode,my system will reboot after about 30 seconds.is this a driver bug?
bug print as follow:

_# [  481.630319] skbuff: skb_over_panic: text:bf06ecbc len:65327 put:359 head:c3170008 data:c3170040 tail:0xc317ff6f end:0xc317ff40 dev:<NULL>
[  481.642699] ------------[ cut here ]------------
[  481.647311] kernel BUG at net/core/skbuff.c:105!
[  481.651921] Internal error: Oops - BUG: 0 [#1] SMP ARM
[  481.657052] Modules linked in: 88XXau(O)
[  481.660983] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G           O    4.13.0-licheepi-zero+ #35
[  481.669492] Hardware name: Allwinner sun8i Family
[  481.674189] task: c0a09680 task.stack: c0a00000
[  481.678726] PC is at skb_panic+0x60/0x64
[  481.682646] LR is at skb_panic+0x60/0x64
[  481.686563] pc : [<c053bf84>]    lr : [<c053bf84>]    psr: 600f0193
[  481.692811] sp : c0a01e18  ip : 00000000  fp : 40000006
[  481.698028] r10: c0a02080  r9 : c431d000  r8 : 0000ff2f
[  481.703245] r7 : c07627b0  r6 : c3170008  r5 : c3170040  r4 : c317ff6f
[  481.709763] r3 : c4000000  r2 : 00000001  r1 : 02000005  r0 : 0000007d
[  481.716282] Flags: nZCv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment none
[  481.723493] Control: 10c5387d  Table: 4311006a  DAC: 00000051
[  481.729230] Process swapper/0 (pid: 0, stack limit = 0xc0a00210)
[  481.735228] Stack: (0xc0a01e18 to 0xc0a02000)
[  481.739580] 1e00:                                                       00000167 c3170008
[  481.747748] 1e20: c3170040 c317ff6f c317ff40 c0823724 c317ff40 c053cec0 c431c000 c431d260
[  481.755916] 1e40: c314a280 c3bdcc48 00000000 bf06ecbc c314a280 600f0113 00000000 c3b2a500
[  481.764083] 1e60: c3b2a4fc c094a284 c0a02080 c0477dac c3b2a4f8 c0a01e80 00000000 c0477eac
[  481.772251] 1e80: c0a01e80 c0a01e80 c3b2a508 c3b2a50c 00000000 c0a01ea0 c0a00000 c012327c
[  481.780418] 1ea0: 00000000 00000006 c0a00000 c0a02098 c0a02080 00000100 c0a02080 c0123434
[  481.788586] 1ec0: c3808000 c4003000 c0a01ec0 c0a424c0 0000000a 000046f3 c0a02d00 00200102
[  481.796754] 1ee0: c3899060 c095160c 0000001a 00000000 00000000 c3808000 c4003000 c3e02d40
[  481.804921] 1f00: 00000000 c0123884 c095160c c0164094 c0a16ce4 c0a04fb4 c400200c c0a01f48
[  481.813089] 1f20: c4002000 c01014a4 c010808c 600f0013 ffffffff c0a01f7c c0a41a40 c0a00000
[  481.821257] 1f40: c3e02d40 c010bdcc 00000000 00045620 0349c000 c01171c0 c0a00000 c0a04d48
[  481.829424] 1f60: c0a04cec c0a3a622 c0a41a40 c0938a28 c3e02d40 00000000 0349c000 c0a01f98
[  481.837592] 1f80: c0108088 c010808c 600f0013 ffffffff 00000051 00000000 c0a00000 c0155a68
[  481.845759] 1fa0: 000000bb c0a41a40 c0a04cc0 ffffffff c0a41a40 c0155d48 c0a41a80 c0900c58
[  481.853927] 1fc0: ffffffff ffffffff 00000000 c0900684 00000000 c0938a28 c0a41dd4 c0a04cd8
[  481.862094] 1fe0: c0938a24 c0a0a8d0 4000406a 410fc075 00000000 4000807c 00000000 00000000
[  481.870281] [<c053bf84>] (skb_panic) from [<c053cec0>] (skb_put+0x50/0x54)
[  481.877706] [<c053cec0>] (skb_put) from [<bf06ecbc>] (usb_read_port_complete+0x9c/0x1d8 [88XXau])
[  481.886896] [<bf06ecbc>] (usb_read_port_complete [88XXau]) from [<c0477dac>] (__usb_hcd_giveback_urb+0x80/0xe4)
[  481.896987] [<c0477dac>] (__usb_hcd_giveback_urb) from [<c0477eac>] (usb_giveback_urb_bh+0x9c/0xec)
[  481.906028] [<c0477eac>] (usb_giveback_urb_bh) from [<c012327c>] (tasklet_action+0x74/0x110)
[  481.914460] [<c012327c>] (tasklet_action) from [<c0123434>] (__do_softirq+0xc8/0x220)
[  481.922282] [<c0123434>] (__do_softirq) from [<c0123884>] (irq_exit+0xd8/0x140)
[  481.929588] [<c0123884>] (irq_exit) from [<c0164094>] (__handle_domain_irq+0x60/0xb4)
[  481.937415] [<c0164094>] (__handle_domain_irq) from [<c01014a4>] (gic_handle_irq+0x4c/0x90)
[  481.945759] [<c01014a4>] (gic_handle_irq) from [<c010bdcc>] (__irq_svc+0x6c/0x90)
[  481.953230] Exception stack(0xc0a01f48 to 0xc0a01f90)
[  481.958276] 1f40:                   00000000 00045620 0349c000 c01171c0 c0a00000 c0a04d48
[  481.966444] 1f60: c0a04cec c0a3a622 c0a41a40 c0938a28 c3e02d40 00000000 0349c000 c0a01f98
[  481.974609] 1f80: c0108088 c010808c 600f0013 ffffffff
[  481.979661] [<c010bdcc>] (__irq_svc) from [<c010808c>] (arch_cpu_idle+0x38/0x3c)
[  481.987055] [<c010808c>] (arch_cpu_idle) from [<c0155a68>] (do_idle+0xc0/0x138)
[  481.994360] [<c0155a68>] (do_idle) from [<c0155d48>] (cpu_startup_entry+0x18/0x1c)
[  482.001926] [<c0155d48>] (cpu_startup_entry) from [<c0900c58>] (start_kernel+0x374/0x380)
[  482.010097] Code: e58d5008 e58d400c e58de010 ebf09d3c (e7f001f2) 
[  482.016186] ---[ end trace 0e11b3253337338a ]---
[  482.020799] Kernel panic - not syncing: Fatal exception in interrupt
[  482.027155] Rebooting in 5 seconds..
_

kimocoder, 2022-06-06:
what branch is this?

lsy512, 2022-06-07:
> what branch is this?

v5.6.4.2
commit 37e27f9165300c89607144b646545fac576ec510

chenmiller1, 2023-02-21:
I met it again in kernel-4.14 when my device startup about 80 seconds.
log:
<1>[   88.507293] Unable to handle kernel NULL pointer dereference at virtual address 00000008
<1>[   88.515604] Mem abort info:
<0>[   88.518461] skbuff: skb_over_panic: text:ffffff800100e130 len:226 put:120 head:ffffffc037625000 data:ffffffc037625e66 tail:0xede end:0xe80 dev:eth0
<4>[   88.518493] ------------[ cut here ]------------
<2>[   88.518496] kernel BUG at net/core/skbuff.c:105!