aircrack-ng/rtl8812au #969: TP-Link Archer TU2 Plus (weird deivce name)

TuxStory, 2022-06-06:
My adaptater is a TP-Link Archer TU2 Plus. 
It was working fine until i saw the code was update and i wanted to reisntall it to see if performance will be beter.
I updated the local git files., uninstall everything and reinstall it.

$ sudo make dkms_install 
Everything seems to be ok.

After a reboot, I can't scan wifi network,i can't do anything. + Strange device name ?!
And light is blinking!

nmcli device status
DEVICE        TYPE      STATE         CONNECTION          
\230H'`1\352  wifi      indisponible  -- 

ip -c a
3: �H'`1�: <BROADCAST,MULTICAST,UP,LOWER_UP>

OS: Linux Mint 20.3 x86_64 
Kernel: 5.4.0-113-generic 

Can you help me ? 
Thanks

elfabx, 2022-06-30:
I could fix the same issue by a small change in include/drv_types_linux.h.

The macro definition:
#define dev_addr_set(netdev, ethdata) _rtw_memcpy(netdev, ethdata, ETH_ALEN)
should become:
#define dev_addr_set(netdev, ethdata) _rtw_memcpy(netdev->dev_addr, ethdata, ETH_ALEN)

Otherwise the uses of this macro are inconsistent with the lines that it is supposed to replace in #961 .

TuxStory, 2022-07-07:
Thanks, it works.

priiduneemre, 2022-07-28:
Thanks a ton @elfabx.

Just encountered the exact same issue.

damien-swarm, 2022-08-04:
I'm on Ubuntu 18.04 / kernel version 5.4.0-96.

I didn't see the same symptoms as OP.

In `journalctl -u NetworkManager` I saw
```
Aug 03 19:07:32 mycomputer NetworkManager[51125]: <info>  [1659578852.6044] device (wlp0s20f0u10): supplicant interface state: inactive -> scanning
Aug 03 19:07:36 mycomputer NetworkManager[51125]: <info>  [1659578856.3057] device (wlp0s20f0u10): supplicant interface state: scanning -> associating
Aug 03 19:07:36 mycomputer NetworkManager[51125]: <info>  [1659578856.4145] device (wlp0s20f0u10): supplicant interface state: associating -> associated
Aug 03 19:07:36 mycomputer NetworkManager[51125]: <info>  [1659578856.4271] device (wlp0s20f0u10): supplicant interface state: associated -> 4-way handshake
Aug 03 19:07:39 mycomputer NetworkManager[51125]: <warn>  [1659578859.4305] sup-iface[0x56321fd6ea40,wlp0s20f0u10]: connection disconnected (reason 15)
Aug 03 19:07:39 mycomputer NetworkManager[51125]: <info>  [1659578859.4307] device (wlp0s20f0u10): supplicant interface state: 4-way handshake -> disconnected
Aug 03 19:07:39 mycomputer NetworkManager[51125]: <info>  [1659578859.4330] device (wlp0s20f0u10): Activation: (wifi) disconnected during association, asking for new key
Aug 03 19:07:39 mycomputer NetworkManager[51125]: <info>  [1659578859.4332] device (wlp0s20f0u10): state change: config -> need-auth (reason 'supplicant-disconnect', sys-iface-state: 'managed')
Aug 03 19:07:39 mycomputer NetworkManager[51125]: <info>  [1659578859.5345] device (wlp0s20f0u10): supplicant interface state: disconnected -> inactive
```

But applying the fix in #989 and rebuilding with `make dkms_install` fixed my problem for some reason, so thanks @elfabx !