aircrack-ng/rtl8812au #972: installation error

crossdefalt, 2022-06-07:
(defalt㉿localhost)-[~/rtl8812au]
└─$ make && make install
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.16.0-kali7-amd64/build M=/home/defalt/rtl8812au  modules
make[1]: *** /lib/modules/5.16.0-kali7-amd64/build: No such file or directory.  Stop.
make: *** [Makefile:2251: modules] Error 2

crossdefalt, 2022-06-07:
# make dkms_install 
cp -r * /usr/src/8812au-5.6.4.2_35491.20191025
dkms add -m 8812au -v 5.6.4.2_35491.20191025
Creating symlink /var/lib/dkms/8812au/5.6.4.2_35491.20191025/source -> /usr/src/8812au-5.6.4.2_35491.20191025
dkms build -m 8812au -v 5.6.4.2_35491.20191025
Error! Your kernel headers for kernel 5.16.0-kali7-amd64 cannot be found at /lib/modules/5.16.0-kali7-amd64/build or /lib/modules/5.16.0-kali7-amd64/source.
You can use the --kernelsourcedir option to tell DKMS where it's located.
make: *** [Makefile:2311: dkms_install] Error 1

sadiqaliero, 2022-09-16:
how did you fix this error

sadiqaliero, 2022-09-16:
I also have the same problem but I see you have resolved it