aircrack-ng/rtl8812au #974: Error when building the driver for kernel '5.4.0-117-generic' on Ubuntu 18.04

pratapkygo, 2022-06-13:
Greetings,

I wanted to use this driver for Archer T2U Plus USB Adapter on Ubuntu 18.04
I have got it to work without any issues previously

However, I am not able to build the driver module for Ubuntu 18.04 specifically for this kernel version

Below is the output for `sudo make dkms_install` :

mkdir: created directory '/usr/src/8812au-5.6.4.2_35491.20191025'
cp -r * /usr/src/8812au-5.6.4.2_35491.20191025
dkms add -m 8812au -v 5.6.4.2_35491.20191025

Creating symlink /var/lib/dkms/8812au/5.6.4.2_35491.20191025/source ->
                 /usr/src/8812au-5.6.4.2_35491.20191025

DKMS: add completed.
dkms build -m 8812au -v 5.6.4.2_35491.20191025

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' -j4 KVER=5.4.0-117-generic KSRC=/lib/modules/5.4.0-117-generic/build...(bad exit status: 2)
ERROR (dkms apport): binary package for 8812au: 5.6.4.2_35491.20191025 not found
Error! Bad return status for module build on kernel: 5.4.0-117-generic (x86_64)
Consult /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/make.log for more information.
Makefile:2308: recipe for target 'dkms_install' failed
make: *** [dkms_install] Error 10

How should I go about to resolve this?
I can provide any other relevant outputs to help debug this

Thanks in advance

danielmoncada, 2022-06-27:
same issue for me, @pratapkygo any solution?

Donnbk, 2022-07-04:
Dear everyone, @pratapkygo @danielmoncada  
I also got this issue. After I searched and tried to fix it, I have just downloaded version 5.3.4 from the link:
https://github.com/aircrack-ng/rtl8812au/tree/v5.3.4
It works well again. 
I hope It also works for you.