aircrack-ng/rtl8812au #975: Cannot be compiled for Linux 5.17.0

wawrzek, 2022-06-18:
I tried to compile the driver with Linux 5.18.6, and it failed. Internet search pointed to the same problem in another driver (https://github.com/tomaspinho/rtl8821ce/pull/267/files)

The fix is applied in my branch: 
https://github.com/wawrzek/rtl8812au/tree/v5.18.3

kimocoder, 2022-06-18:
I tested on 5.17, 5.18 and 5.18 and it's working just fine. And better if you actually attach a compile log/debug/dmesg log

wawrzek, 2022-06-18:
That's what I get, compiling with 5.18.6
```
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.18.3/build M=/home/niewod/System/Kernel/rtl8812au  modules
make[1]: Entering directory '/usr/src/linux-5.18.3'
  CC [M]  /home/niewod/System/Kernel/rtl8812au/os_dep/osdep_service.o
/home/niewod/System/Kernel/rtl8812au/os_dep/osdep_service.c: In function 'thread_exit':
/home/niewod/System/Kernel/rtl8812au/os_dep/osdep_service.c:1312:2: error: implicit declaration of function 'complete_and_exit' [-Werror=implicit-function-declaration]
 1312 |  complete_and_exit(comp, 0);
      |  ^~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:288: /home/niewod/System/Kernel/rtl8812au/os_dep/osdep_service.o] Error 1
make[1]: *** [Makefile:1834: /home/niewod/System/Kernel/rtl8812au] Error 2
make[1]: Leaving directory '/usr/src/linux-5.18.3'
make: *** [Makefile:2514: modules] Error 2
```

The build works fine with the fix. (I don't have Wi-Fi, even if the modules in loaded, but that might be something else.)

CGarces, 2022-06-19:
That compilation issue was fixed moths ago.
https://github.com/aircrack-ng/rtl8812au/pull/942

For some reason you are using the branch v5.13.6 instead of the default one that is maintained and updated.

wawrzek, 2022-06-19:
Oh. The naming of default branch is a bit misleading. I'll take a look later today and close the ticket.

wawrzek, 2022-06-19:
Confirm it works fine with a default branch.

closed 2022-06-19 by wawrzek