aircrack-ng/rtl8812au #976: Stack trace in dmesg when loading driver and replugging device

delbonis, 2022-06-20:
Using tag `v5.6.4.2`, whenever I load the driver with the device plugged in, or replug the device with the driver loaded, I get this error in the output of `dmesg -wH`:

I'm on kernel `5.13.0-48-generic #54-Ubuntu`.  This wasn't happening before a few days ago, it also happens on the `-50-` version of the same Ubuntu kernel release line.

```
[  +0.166922] usbcore: registered new interface driver rtl88XXau
[ +17.588546] usb 1-3: USB disconnect, device number 18
[  +1.611773] usb 1-3: new high-speed USB device number 19 using xhci_hcd
[  +0.236106] usb 1-3: New USB device found, idVendor=2357, idProduct=0101, bcdDevice= 0.00
[  +0.000010] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0.000004] usb 1-3: Product: 802.11n NIC
[  +0.000003] usb 1-3: Manufacturer: Realtek
[  +0.000002] usb 1-3: SerialNumber: 123456
[Jun20 14:19] usb 1-3: 88XXau e8:de:27:a4:c4:a0 hw_info[d7]
[  +0.075183] ------------[ cut here ]------------
[  +0.000004] WARNING: CPU: 1 PID: 12980 at [...snip...]/aircrack-rtl8812au/os_dep/linux/os_intfs.c:3206 rtw_os_ndevs_register+0xbb/0x1a0 [88XXau]
[  +0.000112] Modules linked in: 88XXau(OE) 8821au(OE) rndis_host cdc_ether usbnet mii xt_conntrack nft_chain_nat xt_MASQUERADE nf_nat nf_conntrack_netlink nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 xfrm_user xfrm_algo nft_counter xt_addrtype nft_compat nf_tables libcrc32c nfnetlink br_netfilter bridge stp llc overlay binfmt_misc snd_hda_codec_realtek intel_rapl_msr snd_hda_codec_generic intel_rapl_common nls_iso8859_1 ledtrig_audio snd_hda_codec_hdmi snd_hda_intel snd_intel_dspcfg snd_usb_audio snd_intel_sdw_acpi snd_hda_codec snd_usbmidi_lib snd_hda_core mc snd_hwdep snd_pcm edac_mce_amd amdgpu snd_seq_midi snd_seq_midi_event kvm snd_rawmidi crct10dif_pclmul iommu_v2 ghash_clmulni_intel snd_seq gpu_sched drm_ttm_helper ttm snd_seq_device snd_timer aesni_intel drm_kms_helper crypto_simd cryptd cec rc_core rapl snd cfg80211 i2c_algo_bit fb_sys_fops syscopyarea joydev sysfillrect input_leds wmi_bmof efi_pstore k10temp eeepc_wmi ccp soundcore sysimgblt mac_hid sch_fq_codel ipmi_devintf
[  +0.000090]  ipmi_msghandler msr parport_pc ppdev lp drm parport ip_tables x_tables autofs4 hid_generic mfd_aaeon asus_wmi sparse_keymap usbhid hid video crc32_pclmul nvme i2c_piix4 r8169 ahci gpio_amdpt xhci_pci realtek libahci nvme_core xhci_pci_renesas wmi gpio_generic [last unloaded: 88XXau]
[  +0.000033] CPU: 1 PID: 12980 Comm: kworker/1:0 Tainted: G        W  OE     5.13.0-48-generic #54-Ubuntu
[  +0.000005] Hardware name: System manufacturer System Product Name/PRIME B350-PLUS, BIOS 5602 07/14/2020
[  +0.000003] Workqueue: usb_hub_wq hub_event
[  +0.000007] RIP: 0010:rtw_os_ndevs_register+0xbb/0x1a0 [88XXau]
[  +0.000102] Code: c3 0f 0b 45 31 ed 48 83 c4 08 5b 44 89 e8 41 5c 41 5d 5d c3 48 81 c3 cb 34 00 00 48 c7 c6 78 54 76 c1 3c 01 48 0f 44 f3 eb 9a <0f> 0b 4d 8b ac 24 50 01 00 00 4d 85 ed 0f 84 94 00 00 00 41 80 bd
[  +0.000003] RSP: 0018:ffffbd90823bf728 EFLAGS: 00010297
[  +0.000004] RAX: 0000000000000000 RBX: 0000000000000000 RCX: 0000000000000000
[  +0.000003] RDX: ffffbd90823bf5f8 RSI: ffffbd90823bf5f0 RDI: ffffbd90823bf6b8
[  +0.000002] RBP: ffffbd90823bf748 R08: 0000000000000001 R09: 0000000000000000
[  +0.000002] R10: 0000000000000000 R11: 0000000000000001 R12: ffff959014605000
[  +0.000002] R13: 0000000000000001 R14: ffffffffc176cfc0 R15: ffff95905ccec830
[  +0.000003] FS:  0000000000000000(0000) GS:ffff95920e080000(0000) knlGS:0000000000000000
[  +0.000002] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  +0.000003] CR2: 0000557e185532a0 CR3: 00000001a7910000 CR4: 00000000003506e0
[  +0.000003] Call Trace:
[  +0.000002]  <TASK>
[  +0.000002]  ? rtw_os_ndevs_alloc+0x78/0x100 [88XXau]
[  +0.000109]  rtw_os_ndevs_init+0x2c/0xd0 [88XXau]
[  +0.000107]  rtw_usb_primary_adapter_init+0x5c1/0x8b0 [88XXau]
[  +0.000102]  usb_probe_interface+0xee/0x2a0
[  +0.000006]  really_probe+0x3ac/0x4c0
[  +0.000006]  driver_probe_device+0xf0/0x160
[  +0.000005]  __device_attach_driver+0x79/0xe0
[  +0.000006]  ? driver_allows_async_probing+0x50/0x50
[  +0.000005]  bus_for_each_drv+0x84/0xd0
[  +0.000005]  __device_attach+0xde/0x1f0
[  +0.000005]  device_initial_probe+0x13/0x20
[  +0.000005]  bus_probe_device+0x8f/0xa0
[  +0.000004]  device_add+0x335/0x680
[  +0.000005]  usb_set_configuration+0x45f/0x820
[  +0.000005]  usb_generic_driver_probe+0x53/0x70
[  +0.000005]  usb_probe_device+0x3e/0x100
[  +0.000004]  really_probe+0x3ac/0x4c0
[  +0.000005]  driver_probe_device+0xf0/0x160
[  +0.000005]  __device_attach_driver+0x79/0xe0
[  +0.000005]  ? driver_allows_async_probing+0x50/0x50
[  +0.000005]  bus_for_each_drv+0x84/0xd0
[  +0.000005]  __device_attach+0xde/0x1f0
[  +0.000005]  device_initial_probe+0x13/0x20
[  +0.000004]  bus_probe_device+0x8f/0xa0
[  +0.000005]  device_add+0x335/0x680
[  +0.000004]  usb_new_device.cold+0x11c/0x2f1
[  +0.000007]  hub_port_connect+0x6e9/0x8f0
[  +0.000004]  hub_port_connect_change+0xa0/0x370
[  +0.000004]  port_event+0x5b7/0x700
[  +0.000004]  hub_event+0x1e7/0x430
[  +0.000004]  process_one_work+0x220/0x3c0
[  +0.000007]  worker_thread+0x53/0x420
[  +0.000006]  kthread+0x11f/0x140
[  +0.000004]  ? process_one_work+0x3c0/0x3c0
[  +0.000005]  ? set_kthread_struct+0x50/0x50
[  +0.000005]  ret_from_fork+0x22/0x30
[  +0.000007]  </TASK>
[  +0.000002] ---[ end trace 90ee41601543fdd7 ]---
[  +0.196393] usb 1-3: reset high-speed USB device number 19 using xhci_hcd
```

topeolufe, 2022-06-27:
I'm noticing the same issue as well...

darrylburke, 2022-07-04:
Same here as well.

Dr-Insanity, 2022-07-04:
Oh dear, and I just bought an Alpha AWUS036AC 😳

shidoh, 2022-07-07:
and I bought AWUS036AC 3 days ago :)
and i see the same issue

sagypt, 2022-07-29:
Same here with branch 5.6.4.2 and Alfa Network AWUS036ACH
```
lsb_release -a

No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 11 (bullseye)
Release:	11
Codename:	bullseye
```
```
uname -a

Linux arkham 5.10.0-16-amd64 #1 SMP Debian 5.10.127-2 (2022-07-23) x86_64 GNU/Linux
```
After attaching the NIC
```
lsusb

Bus 002 Device 006: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter
```
```
dmesg -W

[  156.418029] usb 2-2: new high-speed USB device number 6 using xhci_hcd
[  156.566344] usb 2-2: New USB device found, idVendor=0bda, idProduct=8812, bcdDevice= 0.00
[  156.566357] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  156.566363] usb 2-2: Product: 802.11n NIC
[  156.566368] usb 2-2: Manufacturer: Realtek
[  156.566372] usb 2-2: SerialNumber: 123456
[  156.731787] usb 2-2: 88XXau 00:c0:ca:90:08:b5 hw_info[d7]
[  156.765964] ------------[ cut here ]------------
[  156.766017] WARNING: CPU: 1 PID: 1423 at /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/os_dep/linux/os_intfs.c:3206 rtw_os_ndevs_register+0xad/0x190 [88XXau]
[  156.766019] Modules linked in: 88XXau(OE+) rfcomm cmac algif_hash algif_skcipher af_alg vboxnetadp(OE) vboxnetflt(OE) vboxdrv(OE) bnep btusb btrtl btbcm btintel bluetooth intel_rapl_msr intel_rapl_common x86_pkg_temp_thermal intel_powerclamp jitterentropy_rng coretemp kvm_intel kvm drbg irqbypass aes_generic ghash_clmulni_intel rtl8723be snd_hda_codec_conexant snd_hda_codec_generic ledtrig_audio snd_hda_codec_hdmi btcoexist snd_hda_intel rtl8723_common snd_intel_dspcfg rtl_pci soundwire_intel rtlwifi soundwire_generic_allocation uvcvideo aesni_intel crypto_simd snd_soc_core videobuf2_vmalloc mei_hdcp videobuf2_memops snd_compress soundwire_cadence videobuf2_v4l2 videobuf2_common cryptd mac80211 snd_hda_codec glue_helper videodev iTCO_wdt ansi_cprng rtsx_usb_ms rapl snd_hda_core intel_pmc_bxt mc ecdh_generic ecc snd_hwdep intel_cstate iTCO_vendor_support libaes intel_uncore memstick soundwire_bus snd_pcm joydev cfg80211 serio_raw pcspkr sg at24 snd_timer watchdog snd mei_me libarc4
[  156.766073]  soundcore mei ideapad_laptop sparse_keymap rfkill ac evdev binfmt_misc parport_pc ppdev lp parport fuse sunrpc configfs ip_tables x_tables autofs4 ext4 crc16 mbcache jbd2 crc32c_generic hid_logitech_hidpp hid_logitech_dj hid_generic rtsx_usb_sdmmc sd_mod mmc_core sr_mod cdrom t10_pi crc_t10dif crct10dif_generic usbhid rtsx_usb hid i915 nouveau ahci mxm_wmi ttm i2c_algo_bit libahci crct10dif_pclmul drm_kms_helper crct10dif_common libata xhci_pci xhci_hcd ehci_pci r8169 i2c_i801 ehci_hcd realtek mdio_devres crc32_pclmul cec crc32c_intel scsi_mod drm psmouse usbcore i2c_smbus lpc_ich libphy usb_common battery wmi video button
[  156.766122] CPU: 1 PID: 1423 Comm: systemd-udevd Tainted: G           OE     5.10.0-16-amd64 #1 Debian 5.10.127-2
[  156.766123] Hardware name: LENOVO 20354/Lancer 5A5, BIOS 9BCN91WW 07/21/2015
[  156.766159] RIP: 0010:rtw_os_ndevs_register+0xad/0x190 [88XXau]
[  156.766162] Code: 41 5c c3 0f 0b 45 31 e4 48 83 c4 08 44 89 e0 5b 5d 41 5c c3 48 81 c3 cb 34 00 00 48 c7 c6 d8 09 75 c1 3c 01 48 0f 44 f3 eb a0 <0f> 0b 4c 8b a5 50 01 00 00 4d 85 e4 0f 84 9d 00 00 00 41 80 bc 24
[  156.766164] RSP: 0018:ffffbb61c265fbe0 EFLAGS: 00010297
[  156.766166] RAX: 0000000000000000 RBX: 0000000000000000 RCX: 0000000000000000
[  156.766168] RDX: 0000000000000001 RSI: ffffbb61c265fae0 RDI: ffffbb61c265fb90
[  156.766169] RBP: ffff976f93e07000 R08: 7fffffffffffffff R09: 0000000000000000
[  156.766171] R10: ffff9772bf26fdc0 R11: 0000000000000000 R12: 0000000000000001
[  156.766172] R13: ffff976f86126000 R14: ffff976fd04b3000 R15: ffff976f86126000
[  156.766174] FS:  00007fc5b46858c0(0000) GS:ffff9772bf240000(0000) knlGS:0000000000000000
[  156.766176] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  156.766178] CR2: 00007f8bf7976da0 CR3: 0000000152872005 CR4: 00000000001706e0
[  156.766179] Call Trace:
[  156.766215]  ? rtw_os_ndevs_alloc+0x6e/0xf0 [88XXau]
[  156.766247]  rtw_os_ndevs_init+0x25/0xc0 [88XXau]
[  156.766279]  rtw_usb_primary_adapter_init+0x586/0x890 [88XXau]
[  156.766295]  usb_probe_interface+0xe2/0x2a0 [usbcore]
[  156.766301]  really_probe+0x37b/0x480
[  156.766304]  driver_probe_device+0xe1/0x150
[  156.766307]  device_driver_attach+0xa1/0xb0
[  156.766310]  __driver_attach+0x8a/0x150
[  156.766313]  ? device_driver_attach+0xb0/0xb0
[  156.766316]  ? device_driver_attach+0xb0/0xb0
[  156.766318]  bus_for_each_dev+0x78/0xc0
[  156.766321]  bus_add_driver+0x136/0x200
[  156.766325]  driver_register+0x8b/0xe0
[  156.766335]  usb_register_driver+0x84/0x120 [usbcore]
[  156.766337]  ? 0xffffffffc1894000
[  156.766355]  init_module+0x53/0x1000 [88XXau]
[  156.766360]  do_one_initcall+0x44/0x1d0
[  156.766364]  ? do_init_module+0x23/0x240
[  156.766367]  ? kmem_cache_alloc_trace+0xf5/0x200
[  156.766371]  do_init_module+0x4c/0x240
[  156.766374]  __do_sys_finit_module+0xb1/0x110
[  156.766378]  do_syscall_64+0x33/0x80
[  156.766383]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[  156.766385] RIP: 0033:0x7fc5b4b3e9b9
[  156.766388] Code: 00 c3 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 44 00 00 48 89 f8 48 89 f7 48 89 d6 48 89 ca 4d 89 c2 4d 89 c8 4c 8b 4c 24 08 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d a7 54 0c 00 f7 d8 64 89 01 48
[  156.766390] RSP: 002b:00007ffe81c33b68 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
[... cut ...]
[  156.766394] RDX: 0000000000000000 RSI: 00007fc5b4cc9e2d RDI: 0000000000000006
[  156.766396] RBP: 0000000000020000 R08: 0000000000000000 R09: 0000565186925c50
[  156.766397] R10: 0000000000000006 R11: 0000000000000246 R12: 00007fc5b4cc9e2d
[  156.766399] R13: 0000000000000000 R14: 000056518694a950 R15: 000056518696db10
[  156.766402] ---[ end trace 905b3e1902f3f91f ]---
[  156.894390] usb 2-2: reset high-speed USB device number 6 using xhci_hcd
[  157.042987] usbcore: registered new interface driver rtl88XXau
```
NIC doesn't appears in
`lshw -c network`

TexinTX, 2022-08-11:
I get similar on my CentOS 8 VM:Linux osboxes 4.18.0-408.el8.x86_64 #1 SMP Mon Jul 18 17:42:52 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
dmesg:
`[  164.751351] usb 1-2: new high-speed USB device number 3 using xhci_hcd
[  165.008500] usb 1-2: New USB device found, idVendor=0bda, idProduct=a811, bcdDevice= 2.00
[  165.008532] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  165.008546] usb 1-2: Product: 802.11ac WLAN Adapter 
[  165.008558] usb 1-2: Manufacturer: Realtek 
[  165.008569] usb 1-2: SerialNumber: 00e04c000001
[  167.087627] rfkill: input handler disabled
[  168.224792] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[  168.230787] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[  174.862834] hrtimer: interrupt took 2743101 ns
[  175.299980] usb 1-2: 88XXau 16:88:11:c3:6d:b7 hw_info[107]
[  175.450867] WARNING: CPU: 0 PID: 3092 at /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/os_dep/linux/os_intfs.c:3206 rtw_os_ndevs_register+0xa5/0x170 [88XXau]
[  175.451540] Modules linked in: 88XXau(OE+) cfg80211 rfkill nls_utf8 isofs uinput vboxvideo drm_vram_helper drm_ttm_helper xt_CHECKSUM ipt_MASQUERADE xt_conntrack ipt_REJECT nf_nat_tftp nft_objref nf_conntrack_tftp nft_counter bridge stp llc xt_comment nft_compat nft_fib_inet nft_fib_ipv4 nft_fib_ipv6 nft_fib nft_reject_inet nf_reject_ipv4 nf_reject_ipv6 nft_reject nft_ct nf_tables_set nft_chain_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip_set nf_tables nfnetlink tun sunrpc intel_rapl_msr intel_rapl_common intel_pmc_core_pltdrv intel_pmc_core crct10dif_pclmul crc32_pclmul snd_intel8x0 snd_ac97_codec ac97_bus snd_seq snd_seq_device snd_pcm ghash_clmulni_intel rapl snd_timer joydev snd pcspkr soundcore i2c_piix4 vboxguest(OE) video xfs libcrc32c sd_mod sr_mod t10_pi cdrom sg ata_generic vmwgfx ttm drm_kms_helper syscopyarea sysfillrect sysimgblt fb_sys_fops drm ata_piix libata crc32c_intel serio_raw e1000 fuse ipmi_devintf ipmi_msghandler
[  175.452571] CPU: 0 PID: 3092 Comm: systemd-udevd Kdump: loaded Tainted: G           OE    --------- -  - 4.18.0-408.el8.x86_64 #1
[  175.452599] Hardware name: innotek GmbH VirtualBox/VirtualBox, BIOS VirtualBox 12/01/2006
[  175.452609] RIP: 0010:rtw_os_ndevs_register+0xa5/0x170 [88XXau]
[  175.453429] Code: e8 5d 41 5c 41 5d c3 0f 0b 31 ed 5b 89 e8 5d 41 5c 41 5d c3 49 81 c4 8b 35 00 00 48 c7 c6 d0 03 ea c0 3c 01 49 0f 44 f4 eb a6 <0f> 0b 48 8b ab 50 01 00 00 48 85 ed 0f 84 91 00 00 00 80 bd a8 47
[  175.453450] RSP: 0018:ffffa8e540c87af0 EFLAGS: 00010297
[  175.453472] RAX: 0000000000000000 RBX: ffff8ff3c70da000 RCX: 0000000000000000
[  175.453487] RDX: 0000000000000001 RSI: ffffa8e540c87a08 RDI: ffffa8e540c87aa0
[  175.453501] RBP: 0000000000000001 R08: 0000000000000001 R09: 0000000000000d3c
[  175.453514] R10: 000000000000000d R11: 0000000000000000 R12: 0000000000000000
[  175.453525] R13: ffffffffc0eef140 R14: ffffffffc0eef140 R15: ffff8ff3c33be800
[  175.453542] FS:  00007f4d90e1e280(0000) GS:ffff8ff3ffc00000(0000) knlGS:0000000000000000
[  175.453560] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  175.453575] CR2: 00005610c8d500f0 CR3: 00000000084e0001 CR4: 00000000000706f0
[  175.453612] Call Trace:
[  175.453642]  rtw_os_ndevs_init+0x57/0xf0 [88XXau]
[  175.454589]  rtw_usb_primary_adapter_init+0x525/0x870 [88XXau]
[  175.455240]  usb_probe_interface+0xe4/0x280
[  175.455610]  __driver_probe_device+0x23f/0x4c0
[  175.455688]  driver_probe_device+0x1f/0x90
[  175.455714]  __driver_attach+0x77/0x160
[  175.455737]  ? __device_attach_driver+0xf0/0xf0
[  175.455761]  bus_for_each_dev+0x77/0xc0
[  175.455781]  ? klist_add_tail+0x57/0x70
[  175.455834]  bus_add_driver+0x14d/0x1e0
[  175.455860]  driver_register+0x6b/0xb0
[  175.455885]  usb_register_driver+0x88/0x140
[  175.455916]  ? 0xffffffffc0c78000
[  175.455935]  init_module+0x51/0x1000 [88XXau]
[  175.456226]  do_one_initcall+0x46/0x1d0
[  175.456309]  ? do_init_module+0x22/0x220
[  175.456450]  ? kmem_cache_alloc_trace+0x142/0x280
[  175.456506]  do_init_module+0x5a/0x220
[  175.456571]  load_module+0x14ba/0x17f0
[  175.456598]  ? apic_timer_interrupt+0xa/0x20
[  175.456634]  ? __do_sys_init_module+0x13b/0x180
[  175.456652]  __do_sys_init_module+0x13b/0x180
[  175.456678]  do_syscall_64+0x5b/0x1a0
[  175.456707]  entry_SYSCALL_64_after_hwframe+0x65/0xca
[  175.456757] RIP: 0033:0x7f4d8f93d19e
[  175.456782] Code: 48 8b 0d ed 5c 38 00 f7 d8 64 89 01 48 83 c8 ff c3 66 2e 0f 1f 84 00 00 00 00 00 90 f3 0f 1e fa 49 89 ca b8 af 00 00 00 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d ba 5c 38 00 f7 d8 64 89 01 48
[  175.456802] RSP: 002b:00007ffe27d51748 EFLAGS: 00000246 ORIG_RAX: 00000000000000af
[  175.456828] RAX: ffffffffffffffda RBX: 00005610c8d1f560 RCX: 00007f4d8f93d19e
[  175.456844] RDX: 00007f4d9056f86d RSI: 00000000004dd2bd RDI: 00005610c97e5a70
[  175.456857] RBP: 00007f4d9056f86d R08: 00005610c8ca201a R09: 0000000000000000
[  175.456871] R10: 00005610c8ca2010 R11: 0000000000000246 R12: 00005610c97e5a70
[  175.456884] R13: 00005610c8d589e0 R14: 0000000000020000 R15: 0000000000000000
[  175.456907] ---[ end trace b97be9319e343eef ]---
[  175.856698] usb 1-2: reset high-speed USB device number 3 using xhci_hcd
[  176.216965] usbcore: registered new interface driver rtl88XXau
`

lsusb:
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 0bda:a811 Realtek Semiconductor Corp. RTL8811AU 802.11a/b/g/n/ac WLAN Adapter
Bus 001 Device 002: ID 80ee:0021 VirtualBox USB Tablet
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub