aircrack-ng/rtl8812au #977: Raspberry Pi 4B - Driver compiles correctly but does not load properly after boot

mcividino, 2022-06-21:
- Raspberry Pi is running Raspbian Buster 32 bits:
root@arris-probe-1806:~# uname -a
Linux arris-probe-1806 5.10.103-v7l+ #1529 SMP Tue Mar 8 12:24:00 GMT 2022 armv7l GNU/Linux

- Installed the driver following these steps:

$ git clone -b v5.6.4.2 https://github.com/aircrack-ng/rtl8812au.git
cd rtl*
 
sudo apt update --allow-releaseinfo-change
sudo apt-get update
sudo apt-get upgrade
sudo apt-get install build-essential libelf-dev linux-headers
sudo reboot
cd rtl*
sed -i 's/CONFIG_PLATFORM_I386_PC = y/CONFIG_PLATFORM_I386_PC = n/g' Makefile
sed -i 's/CONFIG_PLATFORM_ARM_RPI = n/CONFIG_PLATFORM_ARM_RPI = y/g' Makefile
export ARCH=arm
sed -i 's/^MAKE="/MAKE="ARCH=arm\ /' dkms.conf
make install

- Compilation completed without errors or warnings:
make install
reboot

- After reboot the wlan1 interface doesn't show up:
root@arris-probe-1806:~# ifconfig
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.0.134  netmask 255.255.255.0  broadcast 192.168.0.255
        inet6 2a02:810d:12c0:48a8:8179:a465:5e16:ce64  prefixlen 64  scopeid 0x0<global>
        inet6 fe80::1aa4:bcc1:b302:c218  prefixlen 64  scopeid 0x20<link>
        inet6 2a02:810d:12c0:48a8::76cb  prefixlen 128  scopeid 0x0<global>
        ether dc:a6:32:37:41:04  txqueuelen 1000  (Ethernet)
        RX packets 7374  bytes 1596728 (1.5 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 11928  bytes 5248001 (5.0 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 11094  bytes 5396388 (5.1 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 11094  bytes 5396388 (5.1 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether dc:a6:32:37:41:06  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

root@arris-probe-1806:~# ifconfig
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.0.134  netmask 255.255.255.0  broadcast 192.168.0.255
        inet6 2a02:810d:12c0:48a8:8179:a465:5e16:ce64  prefixlen 64  scopeid 0x0<global>
        inet6 fe80::1aa4:bcc1:b302:c218  prefixlen 64  scopeid 0x20<link>
        inet6 2a02:810d:12c0:48a8::76cb  prefixlen 128  scopeid 0x0<global>
        ether dc:a6:32:37:41:04  txqueuelen 1000  (Ethernet)
        RX packets 7374  bytes 1596728 (1.5 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 11928  bytes 5248001 (5.0 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
[... 135 lines cut ...]
                        status = rtw_os_ndev_register(adapter, name);

                        if (status != _SUCCESS) {
                                **rtw_warn_on(1);**
                                break;
                        }
                }
        }

        if (status != _SUCCESS) {
                for (; i >= 0; i--) {
                        adapter = dvobj->padapters[i];
                        if (adapter)
                                rtw_os_ndev_unregister(adapter);
                }
        }

#if defined(CONFIG_IOCTL_CFG80211)
        if (status != _SUCCESS)
                rtw_cfg80211_dev_res_unregister(dvobj);
#endif
exit:

Any clues on why this is not working will be more than welcome.
Thanks.

Tumult1337, 2022-07-04:
idk a fix for this repo, however, if i use this version https://github.com/lwfinger/rtl8812au it works on my Pi (it runs kali linux)

Mad-Professor, 2022-08-07:
I got my rpi 3B+ out to do some vpn work and updated the pi to 5.15.56-v7+ kernel and got the source using rpi-source. I compiled the driver using aircrack-ng version of rtl8812au and the driver compiles successfully, no errors, did make install but upon reboot and using the device, it looks like it connects but is given invalid ip of 169.254.x.x. after several diagnostic attempts at my wireless ap and dhcp server on the network and at the pi itself; I can't get this thing to get an ip. I connect via ethernet and get ip immediately.  I've used this driver before on 4.19 and worked flawlessly. The device I'm using is edimax ew-7811utc. I'm using raspbian OS buster 32bit.