aircrack-ng/rtl8812au #983: Tp Link Archer Ac600T2U Plus Not detected in ubuntu 20

avichal-a, 2022-07-10:
Tp Link Archer Ac600T2U Plus Not detected

12876.798502] Modules linked in: 88XXau(OE+) rfcomm ccm cmac algif_hash algif_skcipher af_alg bnep nls_iso8859_1 intel_rapl_msr intel_rapl_common snd_hda_codec_hdmi x86_pkg_temp_thermal intel_powerclamp coretemp i915 kvm_intel kvm drm_kms_helper crct10dif_pclmul ghash_clmulni_intel cec rc_core aesni_intel i2c_algo_bit snd_ctl_led mei_hdcp fb_sys_fops syscopyarea snd_hda_codec_realtek snd_hda_codec_generic snd_hda_intel iwlmvm snd_intel_dspcfg rtsx_usb_ms snd_intel_sdw_acpi crypto_simd mac80211 libarc4 cryptd sysfillrect snd_hda_codec snd_hda_core rapl dell_laptop memstick snd_hwdep ledtrig_audio sysimgblt uvcvideo snd_pcm intel_cstate btusb snd_seq_midi btrtl snd_seq_midi_event snd_rawmidi btbcm dell_smm_hwmon btintel snd_seq snd_seq_device dell_wmi input_leds serio_raw bluetooth joydev sparse_keymap ecdh_generic ecc mei_me at24 iwlwifi mei snd_timer snd dell_smbios dcdbas soundcore dell_wmi_descriptor wmi_bmof cfg80211 dell_rbtn acpi_pad mac_hid sch_fq_codel ipmi_devintf ipmi_msghandler
[12876.798656] msr parport_pc ppdev lp parport drm ip_tables x_tables autofs4 rtsx_usb_sdmmc rtsx_usb hid_rmi rmi_core videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common videodev mc r8169 hid_generic crc32_pclmul psmouse ahci xhci_pci i2c_hid_acpi libahci i2c_i801 realtek lpc_ich i2c_smbus xhci_pci_renesas i2c_hid wmi hid video [last unloaded: 88XXau]
[12876.798723] CPU: 2 PID: 20402 Comm: modprobe Tainted: G W OE 5.13.0-52-generic #59~20.04.1-Ubuntu
[12876.798731] Hardware name: Dell Inc. Inspiron 3558/0XCR91, BIOS A07 12/15/2015
[12876.798735] RIP: 0010:rtw_os_ndevs_register+0xbb/0x1a0 [88XXau]
[12876.798908] Code: c3 0f 0b 45 31 ed 48 83 c4 08 5b 44 89 e8 41 5c 41 5d 5d c3 48 81 c3 cb 34 00 00 48 c7 c6 80 a9 26 c1 3c 01 48 0f 44 f3 eb 9a <0f> 0b 4d 8b ac 24 50 01 00 00 4d 85 ed 0f 84 94 00 00 00 41 80 bd
[12876.798915] RSP: 0018:ffffbeb9426379a8 EFLAGS: 00010297
[12876.798921] RAX: 0000000000000000 RBX: 0000000000000000 RCX: 0000000000000000
[12876.798925] RDX: ffffbeb942637878 RSI: ffffbeb942637870 RDI: ffffbeb942637938
[12876.798929] RBP: ffffbeb9426379c8 R08: 0000000000000001 R09: 0000000000000000
[12876.798932] R10: 0000000000000000 R11: 0000000000000000 R12: ffffa07da2da1000
[12876.798936] R13: 0000000000000001 R14: ffffffffc1273320 R15: ffffa07c8beb1430
[12876.798940] FS: 00007f7365c2d540(0000) GS:ffffa07ecb500000(0000) knlGS:0000000000000000
[12876.798945] CS: 0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[12876.798950] CR2: 000055e1702de030 CR3: 0000000223352003 CR4: 00000000003706e0
[12876.798954] Call Trace:
[12876.798958]
[12876.798962] ? rtw_os_ndevs_alloc+0x78/0x100 [88XXau]
[12876.799139] rtw_os_ndevs_init+0x2c/0xd0 [88XXau]
[12876.799311] rtw_usb_primary_adapter_init+0x547/0x8c0 [88XXau]
[12876.799483] usb_probe_interface+0xed/0x290
[12876.799498] really_probe+0x339/0x440
[12876.799507] driver_probe_device+0xf0/0x160
[12876.799514] device_driver_attach+0x5d/0x70
[12876.799522] __driver_attach+0xb2/0x140
[12876.799529] ? device_driver_attach+0x70/0x70
[12876.799536] bus_for_each_dev+0x7e/0xc0
[12876.799548] driver_attach+0x1e/0x20
[12876.799554] bus_add_driver+0x152/0x1f0
[12876.799561] driver_register+0x74/0xd0
[12876.799569] usb_register_driver+0x89/0x130
[12876.799580] init_module+0x57/0x1000 [88XXau]
[12876.799674] ? 0xffffffffc1366000
[12876.799680] do_one_initcall+0x48/0x1d0
[12876.799692] ? __cond_resched+0x19/0x30
[12876.799699] ? kmem_cache_alloc_trace+0x37c/0x440
[12876.799712] do_init_module+0x52/0x240
[12876.799720] load_module+0x125d/0x1440
[12876.799733] __do_sys_finit_module+0xc2/0x120
[12876.799739] ? __do_sys_finit_module+0xc2/0x120
[12876.799749] __x64_sys_finit_module+0x1a/0x20
[12876.799755] do_syscall_64+0x61/0xb0
[12876.799763] ? syscall_exit_to_user_mode+0x27/0x50
[12876.799771] ? __x64_sys_newfstat+0x16/0x20
[12876.799781] ? do_syscall_64+0x6e/0xb0
[12876.799786] ? syscall_exit_to_user_mode+0x27/0x50
[12876.799794] ? __x64_sys_read+0x1a/0x20
[12876.799800] ? do_syscall_64+0x6e/0xb0
[12876.799805] ? do_syscall_64+0x6e/0xb0
[12876.799810] ? asm_exc_page_fault+0x8/0x30
[12876.799821] entry_SYSCALL_64_after_hwframe+0x44/0xae
[12876.799830] RIP: 0033:0x7f7365d6f73d
[12876.799837] Code: 00 c3 66 2e 0f 1f 84 00 00 00 00 00 90 f3 0f 1e fa 48 89 f8 48 89 f7 48 89 d6 48 89 ca 4d 89 c2 4d 89 c8 4c 8b 4c 24 08 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d 23 37 0d 00 f7 d8 64 89 01 48
[12876.799843] RSP: 002b:00007fff74c52c38 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
[12876.799850] RAX: ffffffffffffffda RBX: 000055f8cace0180 RCX: 00007f7365d6f73d
[12876.799854] RDX: 0000000000000000 RSI: 000055f8c9fef358 RDI: 0000000000000003
[12876.799858] RBP: 0000000000040000 R08: 0000000000000000 R09: 0000000000000000
[12876.799862] R10: 0000000000000003 R11: 0000000000000246 R12: 000055f8c9fef358
[12876.799865] R13: 0000000000000000 R14: 000055f8cace0100 R15: 000055f8cace0180
[12876.799872]
[12876.799875] ---[ end trace 73067540b79bd177 ]---
[12876.926567] usb 2-1: reset high-speed USB device number 10 using xhci_hcd
[12877.075260] usbcore: registered new interface driver rtl88XXau
[12879.211510]

lsusb
Bus 001 Device 002: ID 8087:8001 Intel Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 005: ID 0bda:0129 Realtek Semiconductor Corp. RTS5129 Card Reader Controller
Bus 002 Device 004: ID 8087:07dc Intel Corp.
Bus 002 Device 003: ID 0c45:6710 Microdia Integrated Webcam
Bus 002 Device 010: ID 2357:0120 TP-Link 802.11ac WLAN Adapter
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

anwita@anwita-Inspiron-3558:~/monirot/rtl8812au$ sudo dkms status
8812au, 5.6.4.2_35491.20191025, 5.13.0-52-generic, x86_64: installed
rtl88x2bu, 5.3.1: added
rtl88x2bu, 5.8.7.1, 5.13.0-51-generic, x86_64: installed
rtl88x2bu, 5.8.7.1, 5.13.0-52-generic, x86_64: installed

avichal-a, 2022-07-10:
Facing Same error in rpi4

avichal-a, 2022-07-10:
i made git reset up to here
commit f123ff312d65951ddd7e483e7c3ad097abaacfdb (HEAD)
Author: Ice-IX [22827741+Ice-IX@users.noreply.github.com](mailto:22827741+Ice-IX@users.noreply.github.com)
Date: Tue Mar 22 02:03:42 2022 -0400
Compiled and reinserted the driver
worked for me

i think the problem is this patch
commit 5bb4bffa971da0532ffe79f1304c7cd0ad352db8
Author: Shmuel H [shmuelhazan0@gmail.com](mailto:shmuelhazan0@gmail.com)
Date: Sat May 7 14:05:45 2022 +0300

linux: set netdev macaddr using dev_addr_set

dev_addr is const now, therefore, we have to use dev_addr_set
in order to modify it.

error coming from rtw_os_ndevs_register()

koalatux, 2022-07-15:
I had the same issue on Debian bullseye. Going back to commit https://github.com/aircrack-ng/rtl8812au/commit/f123ff312d65951ddd7e483e7c3ad097abaacfdb also worked for me. Thanks @avichal-a

7system7, 2022-10-25:
Same issue. Same solution. Thanks!