aircrack-ng/rtl8812au #985: Compiling error: rtl8812au/core/rtw_br_ext.c:20:11: fatal error: net/ipx.h: No such file or directory

terremoth, 2022-07-14:
```
lucas@ubuntu ~/rtl8812au (v5.6.4.2)$ sudo make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.15.0-41-generic/build M=/home/lucas/rtl8812au  modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-41-generic'
  CC [M]  /home/lucas/rtl8812au/core/rtw_cmd.o
  CC [M]  /home/lucas/rtl8812au/core/rtw_security.o
  CC [M]  /home/lucas/rtl8812au/core/rtw_debug.o
  CC [M]  /home/lucas/rtl8812au/core/rtw_io.o
  CC [M]  /home/lucas/rtl8812au/core/rtw_ioctl_query.o
  CC [M]  /home/lucas/rtl8812au/core/rtw_ioctl_set.o
  CC [M]  /home/lucas/rtl8812au/core/rtw_ieee80211.o
  CC [M]  /home/lucas/rtl8812au/core/rtw_mlme.o
  CC [M]  /home/lucas/rtl8812au/core/rtw_mlme_ext.o
  CC [M]  /home/lucas/rtl8812au/core/rtw_mi.o
  CC [M]  /home/lucas/rtl8812au/core/rtw_wlan_util.o
  CC [M]  /home/lucas/rtl8812au/core/rtw_vht.o
  CC [M]  /home/lucas/rtl8812au/core/rtw_pwrctrl.o
  CC [M]  /home/lucas/rtl8812au/core/rtw_rf.o
  CC [M]  /home/lucas/rtl8812au/core/rtw_chplan.o
  CC [M]  /home/lucas/rtl8812au/core/rtw_recv.o
  CC [M]  /home/lucas/rtl8812au/core/rtw_sta_mgt.o
  CC [M]  /home/lucas/rtl8812au/core/rtw_ap.o
  CC [M]  /home/lucas/rtl8812au/core/mesh/rtw_mesh.o
  CC [M]  /home/lucas/rtl8812au/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /home/lucas/rtl8812au/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /home/lucas/rtl8812au/core/rtw_xmit.o
  CC [M]  /home/lucas/rtl8812au/core/rtw_p2p.o
  CC [M]  /home/lucas/rtl8812au/core/rtw_rson.o
  CC [M]  /home/lucas/rtl8812au/core/rtw_tdls.o
  CC [M]  /home/lucas/rtl8812au/core/rtw_br_ext.o
/home/lucas/rtl8812au/core/rtw_br_ext.c:20:11: fatal error: net/ipx.h: No such file or directory
   20 |  #include <net/ipx.h>
      |           ^~~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:285: /home/lucas/rtl8812au/core/rtw_br_ext.o] Error 1
make[1]: *** [Makefile:1875: /home/lucas/rtl8812au] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-41-generic'
make: *** [Makefile:2244: modules] Error 2
```

terremoth, 2022-07-14:
My OS info:

```
lucas@ubuntuescritorio ~/rtl8812au (v5.6.4.2)$ lsb_release -a
LSB Version:	core-11.1.0ubuntu2-noarch:printing-11.1.0ubuntu2-noarch:security-11.1.0ubuntu2-noarch
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.4 LTS
Release:	20.04
Codename:	focal

lucas@ubuntu ~/rtl8812au (v5.6.4.2)$ uname -a
Linux ubuntu **5.15.0-41-generic** #44~20.04.1-Ubuntu SMP Fri Jun 24 13:27:29 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
```

terremoth, 2022-07-14:
This was started happening this week, last week was "ok"

kimocoder, 2022-07-15:
https://github.com/aircrack-ng/rtl8812au/commit/a72835df07f94439dea74af90c3f726eb3ddf0b7

terremoth, 2022-07-15:
@kimocoder thanks, next week I will try and give the feedback here

terremoth, 2022-07-26:
It worked. Thanks

closed 2022-07-26 by terremoth