aircrack-ng/rtl8812au #986: ../drivers/net/wireless/realtek/rtl8812au/os_dep/linux/ioctl_cfg80211.c:1152:6: error: too many arguments to function call, expected 3, have 8

wezaacyber, 2022-07-18:
when compile nethunter kernel this error appears

CGarces, 2022-08-15:
Please provide full logs

cr4sh-me, 2023-02-18:
> Please provide full logs

  CC      drivers/net/wireless/realtek/rtl8812au/os_dep/linux/usb_intf.o
  CC      drivers/net/wireless/realtek/rtl8812au/os_dep/linux/usb_ops_linux.o
  AR      drivers/tty/serial/built-in.o
  CC      drivers/net/wireless/realtek/rtl8812au/os_dep/linux/ioctl_linux.o
  CC      drivers/tty/vt/consolemap.o
  CC      drivers/staging/qcacld-3.0/core/hdd/src/wlan_hdd_driver_ops.o
  CC      drivers/net/wireless/realtek/rtl8812au/os_dep/linux/xmit_linux.o
  CC      drivers/net/wireless/realtek/rtl8812au/os_dep/linux/mlme_linux.o
  CC      drivers/net/wireless/realtek/rtl8812au/os_dep/linux/recv_linux.o
  CONMK   drivers/tty/vt/consolemap_deftbl.c
  CC      drivers/tty/vt/vt.o
  CC      drivers/staging/qcacld-3.0/core/hdd/src/wlan_hdd_ftm.o
  CC      drivers/staging/qcacld-3.0/core/hdd/src/wlan_hdd_hostapd.o
  CC      drivers/net/wireless/realtek/rtl8812au/os_dep/linux/ioctl_cfg80211.o
  CC      drivers/net/wireless/realtek/rtl8812au/os_dep/linux/wifi_regd.o
  CC      drivers/staging/qcacld-3.0/core/hdd/src/wlan_hdd_ioctl.o
  CC      drivers/net/wireless/realtek/rtl8812au/os_dep/linux/rtw_android.o
  CC      drivers/net/wireless/realtek/rtl8812au/os_dep/linux/rtw_proc.o
  CC      drivers/staging/qcacld-3.0/core/hdd/src/wlan_hdd_main.o
../drivers/net/wireless/realtek/rtl8812au/os_dep/linux/ioctl_cfg80211.c:1152:6: error: too many arguments to function call, expected 3, have 8
                        , pmlmepriv->assoc_req + sizeof(struct rtw_ieee80211_hdr_3addr) + 2
                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../include/net/cfg80211.h:5415:6: note: 'cfg80211_roamed' declared here                                                                                                                                                                     
void cfg80211_roamed(struct net_device *dev, struct cfg80211_roam_info *info,
     ^
1 error generated.                                                                                                                                                                                                                          
make[6]: *** [../scripts/Makefile.build:342: drivers/net/wireless/realtek/rtl8812au/os_dep/linux/ioctl_cfg80211.o] Błąd 1
make[6]: *** Oczekiwanie na niezakończone zadania....
  CC      drivers/staging/qcacld-3.0/core/hdd/src/wlan_hdd_memdump.o
  SHIPPED drivers/tty/vt/defkeymap.c
  CC      drivers/tty/vt/consolemap_deftbl.o
  CC      drivers/tty/vt/defkeymap.o
  CC      drivers/usb/common/common.o
  AR      drivers/tty/vt/built-in.o
  AR      drivers/tty/built-in.o
  CC      drivers/usb/core/usb.o
  CC      drivers/usb/core/hub.o
  CC      drivers/staging/qcacld-3.0/core/hdd/src/wlan_hdd_oemdata.o
  AR      drivers/usb/common/usb-common.o
  AR      drivers/usb/common/built-in.o
  CC      drivers/usb/dwc3/core.o
  CC      drivers/staging/qcacld-3.0/core/hdd/src/wlan_hdd_packet_filter.o
  CC      drivers/staging/qcacld-3.0/core/hdd/src/wlan_hdd_power.o

mousamsangwanXD, 2023-08-26:
I also encountered the same problem while compiling kernel 4.9 for Nethunter.

A quick fix is to just comment out the whole 
```
		#else
		cfg80211_roamed(padapter->pnetdev
			#if LINUX_VERSION_CODE > KERNEL_VERSION(2, 6, 39) || defined(COMPAT_KERNEL_RELEASE)
			, notify_channel
			#endif
			, cur_network->network.MacAddress
			, pmlmepriv->assoc_req + sizeof(struct rtw_ieee80211_hdr_3addr) + 2
			, pmlmepriv->assoc_req_len - sizeof(struct rtw_ieee80211_hdr_3addr) - 2
			, pmlmepriv->assoc_rsp + sizeof(struct rtw_ieee80211_hdr_3addr) + 6
			, pmlmepriv->assoc_rsp_len - sizeof(struct rtw_ieee80211_hdr_3addr) - 6
			, GFP_ATOMIC);
```
It's an old definition of `cfg80211_roamed` used in kernels 2.6.29 and below versions

ziyaniftaliev, 2025-12-01:
Hi. Can you tell me how to build the ALFA AWUS036ACH (RTL8812AU) driver for a Xiaomi Mi A3 running Android 11 LineageOS 18.1 with the official Kali Nethunter custom kernel?