aircrack-ng/rtl8812au #987: TP-Link Archer T9UH v1 [Realtek RTL8814AU] misnamed and -addressed

smurfix, 2022-07-23:
This is w/ 4ab079f
```
$ ip link
…
11: ����: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 2312 qdisc mq state UP mode DEFAULT group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
$ uname -a
Linux v-smurf 5.10.0-16-amd64 #1 SMP Debian 5.10.127-1 (2022-06-30) x86_64 GNU/Linux
```

Any ideas?

jbrockc, 2022-08-04:
I am running current Debian Stable as well and have the same issue. The device name is random characters, and there is no MAC address assigned. Were you able to get the device working at all? If so, what steps did you take? Thank you.

smurfix, 2022-08-04:
I updated to the current kernel from Debian Testing. This is of course not a general solution …

jbrockc, 2022-08-04:
I installed [this](https://github.com/gnab/rtl8812au) using `make`, and it worked after reboot. It is good to know that the newer kernel works as well. Thank you.