aircrack-ng/rtl8812au #993: Unable to build on Arch with kernel 5.19.2

margual56, 2022-08-18:
```C
rtl8812au/os_dep/linux/ioctl_cfg80211.c: At top level:
rtl8812au/os_dep/linux/ioctl_cfg80211.c:10133:20: error: initialization of ‘int (*)(struct wiphy *, struct net_device *, unsigned int)’ from incompatible pointer type ‘int (*)(struct wiphy *, struct net_device *)’ [-Werror=incompatible-pointer-types]
10133 |         .stop_ap = cfg80211_rtw_stop_ap,
      |                    ^~~~~~~~~~~~~~~~~~~~
rtl8812au/os_dep/linux/ioctl_cfg80211.c:10133:20: note: (near initialization for ‘rtw_cfg80211_ops.stop_ap’)
rtl8812au/os_dep/linux/ioctl_cfg80211.c:10219:24: error: initialization of ‘int (*)(struct wiphy *, struct wireless_dev *, unsigned int,  struct cfg80211_chan_def *)’ from incompatible pointer type ‘int (*)(struct wiphy *, struct wireless_dev *, struct cfg80211_chan_def *)’ [-Werror=incompatible-pointer-types]
10219 |         .get_channel = cfg80211_rtw_get_channel,
      |                        ^~~~~~~~~~~~~~~~~~~~~~~~
rtl8812au/os_dep/linux/ioctl_cfg80211.c:10219:24: note: (near initialization for ‘rtw_cfg80211_ops.get_channel’)
rtl8812au/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_wdev_unregister’:
rtl8812au/os_dep/linux/ioctl_cfg80211.c:10398:17: error: ‘struct wireless_dev’ has no member named  current_bss’
10398 |         if (wdev->current_bss) {
      |                 ^~
```
OS: Arch Linux x86_64
Kernel: 5.19.2
Device: RTL8821AU

It worked perfectly with kernel 5.19.1

CGarces, 2022-08-18:
Will be fixed by #992

margual56, 2022-08-18:
It is also fixed on [morrownr's version](https://github.com/morrownr/8821au-20210708).

Thank you! Great job 😄

konsbn, 2022-08-19:
Same issue on my Arch Linux Install. Tried installing with [morrownr's version](https://github.com/morrownr/8821au-20210708). Didn't work. Any ideas when this patch will be merged into this repo?

CGarces, 2022-08-19:
You can use the branch from #992 until wait for merge.

konsbn, 2022-08-20:
> You can use the branch from #992 until wait for merge.

How do I do that?

konsbn, 2022-08-28:
This issue is resolved now.

anglersking, 2024-03-13:
I have fixed this bug ，it work for me .
  you can  change  static int cfg80211_rtw_stop_ap(struct wiphy *wiphy, struct net_device *ndev, unsigned int link_id)    to    static int cfg80211_rtw_stop_ap(struct wiphy *wiphy, struct net_device *ndev)  in  /linux/drivers/net/wireless/realtek/rtl8723ds/os_dep/linux/ioctl_cfg80211.c
[image]