aircrack-ng/rtl8812au #997: do not build on archlinux with kernel 5.19.2

ilyapashuk, 2022-08-21:
hello

after the last update of my archlinux I found that this module don't builds anymore.

my kernel is 5.19.2-arch1-1

my distro is archlinux, I used the aur package "rtl88xxau-aircrack-dkms-git" which uses dkms under the hood.

build log is attached.

[make.log](https://github.com/aircrack-ng/rtl8812au/files/9388934/make.log)

jw-shin1abe, 2022-08-21:
Hi,
I think checking & reviewing process of PR#922 is now in progress. [PR#/992](https://github.com/aircrack-ng/rtl8812au/pull/992) 
BTW, I created tentative workaround patch, by referencing to [morrownr's site](https://github.com/morrownr/8821au-20210708). It worked on my slackware box with kernel-5.19.2.
However, I think it's better to wait marge PR#992.
[rtl8812au-kernel-5.19.2-Fix.patch.gz](https://github.com/aircrack-ng/rtl8812au/files/9388987/rtl8812au-kernel-5.19.2-Fix.patch.gz)

CGarces, 2022-08-21:
Duplicate of #993
ping @iyanmv maybe the archlinux package can be updated without wait for the PR.

buildershed, 2022-08-29:
any update on this?

I've been checking both this and morrownr's and both have seem to have left 5.19.2+ on hold for the past week or two. Fedora has now adopted 5.19.4 so these drivers will no longer work.

CGarces, 2022-08-30:
The isue was fixed by #992