aircrack-ng/rtl8812au #998: unable to load module debian11 kernel 5.10.0-17

legale, 2022-08-22:
Linux debian-xeon 5.10.0-17-amd64 #1 SMP Debian 5.10.136-1 (2022-08-13) x86_64 GNU/Linux

dmesg related part
```
[   59.772645] usb 1-2: reset high-speed USB device number 4 using xhci_hcd
[  217.372109] usb 1-2: USB disconnect, device number 4
[  219.539655] usb 1-2: new high-speed USB device number 5 using xhci_hcd
[  219.872612] usb 1-2: New USB device found, idVendor=0bda, idProduct=8812, bcdDevice= 0.00
[  219.872614] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  219.872615] usb 1-2: Product: 802.11n NIC
[  219.872616] usb 1-2: Manufacturer: Realtek
[  219.872617] usb 1-2: SerialNumber: 123456
[  220.190914] usb 1-2: 88XXau 00:c0:ca:a6:37:e6 hw_info[d7]
[  220.223072] ------------[ cut here ]------------
[  220.223111] WARNING: CPU: 8 PID: 131 at /home/ru/rtl8812au/os_dep/linux/os_intfs.c:3206 rtw_os_ndevs_register+0xb5/0x1a0 [88XXau]
[  220.223112] Modules linked in: intel_rapl_msr intel_rapl_common kvm_intel kvm irqbypass ghash_clmulni_intel aesni_intel libaes crypto_simd cryptd glue_helper rapl 88XXau(OE) nls_ascii iTCO_wdt nls_cp437 intel_pmc_bxt vfat iTCO_vendor_support snd_intel8x0 watchdog vmwgfx fat cfg80211 snd_ac97_codec ttm ac97_bus snd_pcm rfkill drm_kms_helper snd_timer joydev sg snd serio_raw pcspkr efi_pstore cec evdev soundcore rng_core vboxguest ac button drm fuse configfs efivarfs ip_tables x_tables autofs4 ext4 crc16 mbcache jbd2 crc32c_generic hid_generic usbhid hid sd_mod t10_pi crc_t10dif crct10dif_generic sr_mod cdrom ahci xhci_pci ata_generic libahci ata_piix xhci_hcd libata usbcore scsi_mod crct10dif_pclmul psmouse crct10dif_common crc32_pclmul crc32c_intel e1000 lpc_ich i2c_piix4 video usb_common
[  220.223159] CPU: 8 PID: 131 Comm: kworker/8:1 Tainted: G        W  OE     5.10.0-17-amd64 #1 Debian 5.10.136-1
[  220.223160] Hardware name: innotek GmbH VirtualBox/VirtualBox, BIOS VirtualBox 12/01/2006
[  220.223172] Workqueue: usb_hub_wq hub_event [usbcore]
[  220.223194] RIP: 0010:rtw_os_ndevs_register+0xb5/0x1a0 [88XXau]
[  220.223196] Code: 0b 45 31 e4 48 83 c4 08 44 89 e0 5b 5d 41 5c c3 cc cc cc cc 48 81 c3 cb 34 00 00 48 c7 c6 c8 36 dd c0 3c 01 48 0f 44 f3 eb 98 <0f> 0b 4c 8b a5 50 01 00 00 4d 85 e4 0f 84 9d 00 00 00 41 80 bc 24
[  220.223197] RSP: 0018:ffff9d1b403f3958 EFLAGS: 00010297
[  220.223198] RAX: 0000000000000000 RBX: 0000000000000000 RCX: 0000000000000000
[  220.223199] RDX: 0000000000000001 RSI: ffff9d1b403f3858 RDI: ffff9d1b403f3908
[  220.223200] RBP: ffff895343617000 R08: 7fffffffffffffff R09: 0000000000000000
[  220.223201] R10: ffff895a1462fdc0 R11: 0000000000000000 R12: 0000000000000001
[  220.223202] R13: ffff895343105800 R14: ffff895340b02000 R15: ffff895343105800
[  220.223203] FS:  0000000000000000(0000) GS:ffff895a14600000(0000) knlGS:0000000000000000
[  220.223204] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  220.223205] CR2: 0000560756d61a78 CR3: 000000012115a003 CR4: 00000000000706e0
[  220.223209] Call Trace:
[  220.223232]  ? rtw_os_ndevs_alloc+0x76/0x100 [88XXau]
[  220.223254]  rtw_os_ndevs_init+0x29/0xc0 [88XXau]
[  220.223276]  rtw_usb_primary_adapter_init+0x58a/0x8a0 [88XXau]
[  220.223284]  usb_probe_interface+0xe2/0x2a0 [usbcore]
[  220.223289]  really_probe+0x37f/0x480
[  220.223291]  driver_probe_device+0xe5/0x150
[  220.223293]  ? driver_allows_async_probing+0x50/0x50
[  220.223294]  bus_for_each_drv+0x7e/0xd0
[  220.223296]  __device_attach+0xbc/0x210
[  220.223298]  bus_probe_device+0x8e/0xb0
[  220.223300]  device_add+0x39d/0x840
[  220.223302]  ? _cond_resched+0x16/0x50
[  220.223308]  usb_set_configuration+0x46f/0x830 [usbcore]
[  220.223315]  usb_generic_driver_probe+0x50/0x70 [usbcore]
[  220.223321]  usb_probe_device+0x39/0x100 [usbcore]
[  220.223323]  really_probe+0x37f/0x480
[  220.223325]  driver_probe_device+0xe5/0x150
[  220.223327]  ? driver_allows_async_probing+0x50/0x50
[  220.223328]  bus_for_each_drv+0x7e/0xd0
[  220.223330]  __device_attach+0xbc/0x210
[  220.223331]  bus_probe_device+0x8e/0xb0
[  220.223333]  device_add+0x39d/0x840
[  220.223339]  usb_new_device.cold+0x111/0x2d3 [usbcore]
[  220.223345]  hub_event+0x1552/0x1850 [usbcore]
[  220.223350]  process_one_work+0x1b6/0x350
[  220.223352]  worker_thread+0x53/0x3e0
[  220.223354]  ? process_one_work+0x350/0x350
[  220.223355]  kthread+0x11b/0x140
[  220.223357]  ? __kthread_bind_mask+0x60/0x60
[  220.223360]  ret_from_fork+0x22/0x30
[  220.223362] ---[ end trace c7c8e4ae0b820e38 ]---
[  220.535632] usb 1-2: reset high-speed USB device number 5 using xhci_hcd
```

TexinTX, 2022-09-06:
> Hello! I got a problem. Unable to load a module.
> 
> uname -a Linux debian-xeon 5.10.0-17-amd64 #1 SMP Debian 5.10.136-1 (2022-08-13) x86_64 GNU/Linux
> 
> dmesg related part

Is this the same issue as?:
https://github.com/aircrack-ng/rtl8812au/issues/976
Stack trace in dmesg when loading driver and replugging device #976

legale, 2022-09-07:
> > Hello! I got a problem. Unable to load a module.
> > uname -a Linux debian-xeon 5.10.0-17-amd64 #1 SMP Debian 5.10.136-1 (2022-08-13) x86_64 GNU/Linux
> > dmesg related part
> 
> Is this the same issue as?: #976 Stack trace in dmesg when loading driver and replugging device #976

Very likely it is.

Jibun-no-Kage, 2022-10-09:
I got a kernel tainting issue, but may be similar root cause?

[ 4750.809661] ------------[ cut here ]------------
[ 4750.809825] WARNING: CPU: 0 PID: 9012 at net/wireless/sme.c:757 __cfg80211_connect_result+0x398/0x400 [cfg80211]
[ 4750.809933] Modules linked in: 88XXau(OE) cfg80211 rfkill 8021q garp stp mrp llc snd_intel8x0 snd_ac97_codec i915 ac97_bus snd_pcm snd_timer snd soundcore pcspkr serio_raw iTCO_wdt video intel_pmc_bxt drm_kms_helper joydev iTCO_vendor_support evdev cec watchdog i2c_algo_bit sg button rng_core bonding parport_pc ppdev lp parport fuse drm configfs ip_tables x_tables autofs4 ext4 crc16 mbcache jbd2 crc32c_generic hid_generic uas usbhid hid usb_storage sd_mod t10_pi sr_mod crc_t10dif cdrom crct10dif_generic crct10dif_common uhci_hcd ehci_pci ata_generic ehci_hcd ata_piix libata psmouse scsi_mod usbcore i2c_i801 lpc_ich e1000 i2c_smbus e100 mii usb_common floppy
[ 4750.810636] CPU: 0 PID: 9012 Comm: kworker/u4:1 Tainted: G           OE     5.10.0-18-686-pae #1 Debian 5.10.140-1
[ 4750.810744] Hardware name: Compaq Evo D510 SFF/07E4h, BIOS 686O2 v3.18 05/28/2003
[ 4750.810844] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
[ 4750.810934] EIP: __cfg80211_connect_result+0x398/0x400 [cfg80211]
[ 4750.811005] Code: 84 81 fd ff ff 3e ff 40 ec e9 78 fd ff ff 8d 74 26 00 0f 0b 8b 56 08 e8 a6 64 fd ff e9 2f fe ff ff 90 0f 0b 8d b6 00 00 00 00 <0f> 0b e9 1f fe ff ff 90 e8 2b 1a 1b cd e9 14 fe ff ff 8d b6 00 00
[ 4750.811189] EAX: 00000000 EBX: c2158000 ECX: c2aa7644 EDX: 00000000
[ 4750.811261] ESI: c2aa760c EDI: 00000000 EBP: c20ddf08 ESP: c20ddee4
[ 4750.811332] DS: 007b ES: 007b FS: 00d8 GS: 00e0 SS: 0068 EFLAGS: 00010246
[ 4750.811410] CR0: 80050033 CR2: 005b7f50 CR3: 01fe6000 CR4: 000006f0
[ 4750.811511] Call Trace:
[ 4750.811574]  cfg80211_process_wdev_events+0x12d/0x1a0 [cfg80211]
[ 4750.811712]  cfg80211_process_rdev_events+0x37/0x80 [cfg80211]
[ 4750.811796]  cfg80211_event_work+0x1b/0x30 [cfg80211]
[ 4750.811862]  process_one_work+0x16c/0x2e0
[ 4750.811916]  worker_thread+0x15e/0x3e0
[ 4750.811968]  kthread+0xf6/0x110
[ 4750.812011]  ? process_one_work+0x2e0/0x2e0
[ 4750.812061]  ? kthread_associate_blkcg+0xb0/0xb0
[ 4750.812121]  ret_from_fork+0x1c/0x28
[ 4750.812169] ---[ end trace b82b89e5cf032eb7 ]---