aircrack-ng/rtl8812au #999: Compilation failing on Rocky/RHEL

dipjyoti-skanray, 2022-08-25:
Any leads on the below ? Compilation failing on Rocky 8.5 !!

CC [M]  xxxxxxx/Downloads/rtl8812au-5.6.4.2/os_dep/linux/ioctl_cfg80211.o
xxxxxxx/Downloads/rtl8812au-5.6.4.2/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_ch_switch_notify’:
xxxxxxx/Downloads/rtl8812au-5.6.4.2/os_dep/linux/ioctl_cfg80211.c:466:3: error: too few arguments to function ‘cfg80211_ch_switch_started_notify’
   cfg80211_ch_switch_started_notify(adapter->pnetdev, &chdef, 0);
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from xxxxxxx/Downloads/rtl8812au-5.6.4.2/include/osdep_service_linux.h:97,
                 from xxxxxxx/Downloads/rtl8812au-5.6.4.2/include/osdep_service.h:58,
                 from xxxxxxx/Downloads/rtl8812au-5.6.4.2/include/drv_types.h:26,
                 from xxxxxxx/Downloads/rtl8812au-5.6.4.2/os_dep/linux/ioctl_cfg80211.c:17:
./include/net/cfg80211.h:7673:6: note: declared here
 void cfg80211_ch_switch_started_notify(struct net_device *dev,
      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
xxxxxxx/Downloads/rtl8812au-5.6.4.2/os_dep/linux/ioctl_cfg80211.c: At top level:
xxxxxxx/Downloads/rtl8812au-5.6.4.2/os_dep/linux/ioctl_cfg80211.c:10193:3: error: ‘struct cfg80211_ops’ has no member named ‘mgmt_frame_register’
  .mgmt_frame_register = cfg80211_rtw_mgmt_frame_register,
   ^~~~~~~~~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:316: xxxxxxx/Downloads/rtl8812au-5.6.4.2/os_dep/linux/ioctl_cfg80211.o] Error 1
make[1]: *** [Makefile:1571: _module_xxxxxxx/Downloads/rtl8812au-5.6.4.2] Error 2
make[1]: Leaving directory '/usr/src/kernels/4.18.0-348.20.1.el8_5.x86_64'

Using kernel 4.18.0-348.20.1.el8_5.x86_64..

CGarces, 2022-08-25:
Duplicate of #873 #780 #542

RHEL has made some backports to old kernels that made complicated to maintain this drivers

Please check issue #861
And this patch
https://github.com/aircrack-ng/rtl8812au/issues/861#issuecomment-859794427

Also you must change  #811 with a similar patch.

But please consider using a different kernel as suggested at
https://github.com/aircrack-ng/rtl8812au/issues/542#issuecomment-587775745

kimocoder, 2023-02-03:
This issue should be fixed

closed 2023-02-03 by kimocoder