aircrack-ng/rtl8812au #1000: Makefile:2266: install Error 1

rightrice, 2022-08-28:
Hi,
I am receiving this error when trying to do `make && make install`:
```
install -p -m 644 88XXau.ko  /lib/modules/5.18.0-kali5-amd64/kernel/drivers/net/wireless/
install: cannot create regular file '/lib/modules/5.18.0-kali5-amd64/kernel/drivers/net/wireless/88XXau.ko': Permission denied
make: *** [Makefile:2266: install] Error 1
```
I have also tried using sudo given it says permission denied. Not sure what the problem could be.

TrickishComic91, 2022-09-01:
I have the same issue, no clue what's causing it.

GyouzaRu, 2022-09-23:
I solved this problem by input `sudo make && sudo make install`.
Maybe it can help you.

leftsideanalog, 2022-10-16:
> I solved this problem by input `sudo make && sudo make install`. Maybe it can help you.

Using POP! OS this helped