aircrack-ng/rtl8812au #1003: Cannot install driver on Manjaro Linux

homie-desktop, 2022-09-26:
Hi. I'm having issues with installing the driver on Manjaro. I've previously installed the driver without any issue on Ubuntu 22.04 and Pop!_OS 22.04. However, in Manjaro the same installation process does not work. Please help, Thanks!

Here's my system information

```
~/rtl8812au (v5.6.4.2 ✔) uname -r
5.15.65-1-MANJARO

~/rtl8812au (v5.6.4.2 ✔) cat /etc/os-release
NAME="Manjaro Linux"
ID=manjaro
ID_LIKE=arch
BUILD_ID=rolling
PRETTY_NAME="Manjaro Linux"
```

Here's the output from `sudo make dkms_install`
```
mkdir: created directory '/usr/src/8812au-5.6.4.2_35491.20191025'
cp -r * /usr/src/8812au-5.6.4.2_35491.20191025
dkms add -m 8812au -v 5.6.4.2_35491.20191025
Creating symlink /var/lib/dkms/8812au/5.6.4.2_35491.20191025/source -> /usr/src/8812au-5.6.4.2_35491.20191025
dkms build -m 8812au -v 5.6.4.2_35491.20191025
Sign command: /usr/lib/modules/5.15.65-1-MANJARO/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Binary sign-file not found, module won't be signed
Error! Your kernel headers for kernel 5.15.65-1-MANJARO cannot be found at /usr/lib/modules/5.15.65-1-MANJARO/build or /usr/lib/modules/5.15.65-1-MANJARO/source.
Please install the linux-headers-5.15.65-1-MANJARO package or use the --kernelsourcedir option to tell DKMS where it's located.
make: *** [Makefile:2317: dkms_install] Error 1
```

Here's the output from `sudo make && sudo make install`
```
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.15.65-1-MANJARO/build M=/home/manjaro_os/rtl8812au  modules
make[1]: *** /lib/modules/5.15.65-1-MANJARO/build: No such file or directory.  Stop.
make: *** [Makefile:2257: modules] Error 2
install -p -m 644 88XXau.ko  /lib/modules/5.15.65-1-MANJARO/kernel/drivers/net/wireless/
install: cannot stat '88XXau.ko': No such file or directory
make: *** [Makefile:2266: install] Error 1
```

Frostie314159, 2022-11-14:
You need to install the kernel headers.
```
sudo pacman -Slinux515-headers
```

homie-desktop, 2022-11-14:
Hi. Thanks! I got it working now. It was suggested to me in the Manjaro forum as well. I forgot to close the issue here. Thanks anyways!

closed 2022-11-14 by homie-desktop