aircrack-ng/rtl8812au #1015: `rtw_switch_usb_mode=1` to make a dongle use USB-3 seems to cripple 2.4GHz reception,

ipaqmaster, 2022-11-05:
Hi,

I recently purchased the `NETGEAR Nighthawk AC1900 USB 3.0 Dual Band adapter` to use on my Dell XPS 13 9310.

The adapter performs outstandingly better than the laptop's inbuilt PCIe Intel AX201 Wi-Fi 6 wifi chip. In an `iw scan` comparison it detects more than double the nearby unique APs BSSIDs than the laptop's onboard wifi can (70 vs 150!)

This dongle uses rtl88XXau which I've built from here. It works OK in USB2 mode and appropriately shows up under `lsusb -t` as 480M speed indicating it is definitely in that mode.

The problem I'm experiencing is when I ask it to enter USB3 mode by setting `rtw_switch_usb_mode=1` on the driver.

You can do this live by either echoing `1` into the parameter of the driver and replugging the usb wifi dongle, or have it more permanently by adding `options 88XXau rtw_switch_usb_mode=1` to a conf file in /etc/modprobe.d and reprobing the driver.

----------

Straight to the point

Using airodump-ng with the dongle in USB3 mode it still fails to hear "nearly" anything in the 2.4GHz range. If I turn on my phone's hotspot in "compatibility mode" (2.4GHz) it does see that, and the phone is right next to me.

Yet if I set the driver back to USB2 mode, replug the dongle and use airodump again it quickly finds over 40 ESSIDs in just a few seconds right now all in the 2.4GHz range, including my phone's hotspot of course.

In both approaches the dongle appears under `lsusb -t` as a USB3 device mentioning 5000M at the end of it, indicating it's in USB3 mode and it's parent and root USB hubs are both also capable of 5000M and 10000M respectively.. so I do not believe the host is to blame here

I guess my question here is... why does this dongle's USB3 mode seem to cripple the device's 2.4GHz hearing ability? It's like it's not even trying when in this mode... but it can still sweep 5GHz channels just fine in USB3 mode... just not 2.4GHz.

Even my tiny thumbnail sized USB2 802.11n 2.4GHz dongle in my bag is severely outperforming monitoring and capturing than this expensive new dongle when it enters USB3 mode. Makes me wonder if this dongle can actually deliver the theoretical 1300Mbps it offers if associated to a capable 5GHz access point.

I'd love any help on the problem or guidance on or guidance on this rtw_switch_usb_mode parameter's usage and performance struggles.

------

To guess and ramble... Maybe it was never supposed to be human controlled and is automatically flipped to when the card associates to a network needing USB3 speeds? (Which would never be any 2.4GHz channel if you get what I mean). But that is only a guess. I'd rather the device just function in USB3 mode at all times if possible and still have great reception on all 2.4+5GHz bands.

ipaqmaster, 2022-11-06:
Looks like I'm not alone nor the first experiencing strangeness when it comes to the driver's USB3 mode and spotty 2.4GHz coverage while it's active.

1. https://github.com/aircrack-ng/rtl8814au/issues/1

>USB 3.0 partially works, .. but, in USB 3.0 mode, it only shows the two networks closest to me, .. does not see the rest of the networks (with usb.2.0 it sees 20+ networks ...) ... some problems with signal strength, apparently ..

2. https://github.com/aircrack-ng/rtl8812au/issues/901 (Possibly same cause)

3. https://github.com/aircrack-ng/rtl8812au/issues/77 with a lot of similar sounding chatter but seems barely relevant to this scenario.

ipaqmaster, 2022-11-11:
Some highly unscientific research on my side but some experiments and thoughts:

This testing is on my `0846:9054 NetGear, Inc. Nighthawk A7000 802.11ac Wireless Adapter AC1900 [Realtek 8814AU]` USB WiFi card.

Works completely fine in USB2 mode, no complaints in any single way on this driver. None. (Yet)

USB3 mode however, **works entirely fine in 5GHz ranges** but "basically" not at all in the 2.4GHz space, with heavy exception. For example:

1. Setting the 88XXau driver's rtw_switch_usb_mode parameter to `1` and connecting the dongle works for USB3 mode, verifiable with `lsusb -t` showing it as a 5000M device (Keep an eye out that the parent device is also 5000M, 10000M or greater for the true USB 3.0 experience)

2. Setting the device to monitor mode as usual and using the classic `airodump-ng` with no extra args to scan the 2.4GHz range finds..... nothing. No APs.

3. If I turn on my iPhone's hotspot and check "Compatibility mode" (To force it to use 2.4G), the dongle can see this. Not too surprising as my phone is right here on the same desk. But the dongle does not see beacons from the other 20+ 2.4GHz APs nearby

This is using a very short 15cm braided Alogic USB-C>USB-A cable as my Dell XPS 13 here only takes USB-C.

-----------

Something that really started to capture my attention was [this paper from Intel regarding noise from the USB 3.0 standard and how it's (You guessed it) output in the 2.4-2.5GHz spectrum.
](https://www.usb.org/sites/default/files/327216.pdf)

So in a pathetic attempt and lack of better insulating material at hand... I tried grasping this tiny cable in my hands and the base of the USB adapter to (Very poorly) isolate it...and it worked. Kind of. The beacons for about 9 nearby BSSIDs started showing up which is not nearly as many as there are nearby, but a significant (infinite) improvement over the zero it can usually see (excluding my phone right here). Then I let go and the beacon counter stopped going up.

This leaves me thinking if NETGEAR really thought this thing deserved a USB3 while shipping it with a long magnetic USB3 extension cradle... they must have known 2.4G performance would've been horrible if it actually negotiates into USB3 mode as it seems USB3 and 2.4G on this unit don't seem to mix, not even a little.

I can only assume the `rtw_switch_usb_mode` is reserved explicitly for when dealing with 5GHz networks to take advantage of potential throughput if you somehow pull off a 40/80MHz wide channel without neighbor noise on your AP. (Only a best guess)

Unfortunately, [this paper from Intel](https://www.usb.org/sites/default/files/327216.pdf) seems to at least partially confirm why my poor-mans arm-cover test seemed to yield some results confirming that USB3 does interfere with the 2.4G range and in fact, very badly. At least with this dongle's design even when fully extended.

With that in mind, I can only assume USB3 mode is only intended for working with 5GHz activity and maybe the official driver switches on demand, or something.

---------------

TL;DR: https://www.usb.org/sites/default/files/327216.pdf the poor 2.4GHz reception on my USB Netgear A7000 could be expected in 88XXau's USB3 mode.

I'd love to try this again with a more expensive highly isolated USB3 cable or directly into a desktop motherboard to compare how it handles the noise for 2.4G reception

It's also likely that other USB3 wifi dongle manufactures know of this issue and may have isolated/designed their onboard components and antennas differently to help compensate. Not sure. Might be worth modding my A7000 here to help further isolate this later to see if I can avoid flipping between USB2/3 mode and just leave it in 3. Could be fun.

If this could still be a software level issue that doesn't involve just falling back to USB2 mode and "Dealing with it". I'd love to hear other experiences with other monitor-mode-capable USB3 WiFi dongles.

closed 2022-11-11 by ipaqmaster

needUpgrades, 2024-05-15:
Yeah you're totally right about the usb 2 and 3 modes, it's not only with rtl drivers

fa1ke5, 2025-06-01:
USB3.0 вызывает интерференцию на WIFI устройствах в полосе 2.4Ghz, поэтому для диапазона 2.4 нужно использовать USB2.0, а для 5.8 - USB3.x