aircrack-ng/rtl8812au #1025: 8812au=8812bu

LLH-l, 2022-12-03:
was found through experiments that,
8812au driver seems to be compatible with 8812bu

[image]
[image]

TheCustomFHD, 2023-01-05:
I am interested on how you managed to load this driver on the rtl8812bu variant, as i also have a TP-Link Archer T4U v3, with this chipset.. did you just rename the kernel driver file?

LLH-l, 2023-01-12:
> I am interested on how you managed to load this driver on the rtl8812bu variant, as i also have a TP-Link Archer T4U v3, with this chipset.. did you just rename the kernel driver file?

Use wifislax64-3.2-final.iso
It already contains 8812bu driver

closed 2024-03-01 by LLH-l