aircrack-ng/rtl8812au #1036: CFI Kernel Panic on Kernel v5.4.147

seigtm, 2022-12-28:
Always get kernel panic from CFI with this driver when plugging in a Wi-Fi device.
There is no way for me to disable CFI flags because my target device's security system depends on and won't boot w/o them.

**Unavoidable config flags**:
```
CONFIG_LTO=y
CONFIG_ARCH_SUPPORTS_LTO_CLANG=y
CONFIG_ARCH_SUPPORTS_THINLTO=y
CONFIG_THINLTO=y
# CONFIG_LTO_NONE is not set
CONFIG_LTO_CLANG=y
CONFIG_CFI_CLANG=y
CONFIG_CFI_CLANG_SHADOW=y
# CONFIG_CFI_PERMISSIVE is not set
```

**CFI failures occur either from `usb_recv_tasklet()` or `rtw_xmit_entry()`.**

**last_dkms №1**:
```
Kernel panic - not syncing: CFI failure (target: usb_recv_tasklet.cfi_jt+0x0/0x4)
CPU: 0 PID: 4319 Comm: Thread-7 Tainted: G S      W  O      5.4.147-qgki #1
Call trace:
dump_backtrace.cfi_jt+0x0/0x4
show_stack+0x18/0x24
dump_stack+0xb8/0x114
panic+0x1b0/0x428
__ubsan_handle_cfi_check_fail_abort+0x0/0x14
perf_proc_update_handler+0x0/0xcc
__cfi_check+0x7cefc/0x7dfb0
tasklet_action_common+0x388/0x41c
tasklet_action+0x24/0x30
__do_softirq+0x20c/0x48c
irq_exit+0xbc/0xc0
__handle_domain_irq+0xa8/0x100
gic_handle_irq+0xcc/0x168
el0_irq_naked+0x58/0x64
SMP: stopping secondary CPUs
```

**last_dkms №2**:
```
Kernel panic - not syncing: CFI failure (target: rtw_xmit_entry.cfi_jt+0x0/0x4)
CPU: 4 PID: 236 Comm: kworker/4:2 Tainted: G S      W  O      5.4.147-qgki #1
Workqueue: ipv6_addrconf addrconf_dad_work
Call trace:
dump_backtrace.cfi_jt+0x0/0x4
show_stack+0x18/0x24
dump_stack+0xb8/0x114
panic+0x1b0/0x428
__ubsan_handle_cfi_check_fail_abort+0x0/0x14
perf_proc_update_handler+0x0/0xcc
__cfi_check+0x7cefc/0x7dfb0
xmit_one+0x2ac/0x2b0
dev_hard_start_xmit+0x58/0xa4
sch_direct_xmit+0x114/0x3b8
__qdisc_run+0x88/0xe0
__dev_xmit_skb+0x15c/0x440
__dev_queue_xmit.llvm.461251673237399726+0x374/0x6b8
neigh_resolve_output+0x198/0x20c
ip6_finish_output2+0x554/0x830
__ip6_finish_output+0x204/0x2ac
ip6_finish_output+0x54/0xe4
ip6_output+0xcc/0x1c4
ndisc_send_skb+0x368/0x550
ndisc_send_ns+0x25c/0x298
addrconf_dad_work+0x3b8/0x6a4
process_one_work+0x2d8/0x574
worker_thread+0x28c/0x518
kthread+0x16c/0x17c
ret_from_fork+0x10/0x18
SMP: stopping secondary CPUs
```

Any help would be appreciated 🥺

GeorgeBannister, 2023-01-23:
I'm having the same issue - Trying to up the devices causes kernel panic due to CFI

GeorgeBannister, 2023-01-25:
This made it work for me (Note I was only having issues with usb_recv_tasklet - you may need to fix signatures for rtw_xmit_entry)
https://github.com/aircrack-ng/rtl8812au/pull/1041

GeorgeBannister, 2023-01-25:
I don't get the rtw_xmit_entry issue when doing this from the README:

  2. Set interface down
  ```
  $ sudo ip link set wlan0 down
  ```
  3. Set monitor mode
  ```
  $ sudo iw dev wlan0 set type monitor
  ```
  4. Set interface up
  ```
  $ sudo ip link set wlan0 up
  ```

seigtm, 2023-01-25:
And, yes, I've already made some tests with this fix and everything works fine as it should.