aircrack-ng/rtl8812au #1038: NetworkManager cannot connect to wifi and crashes the whole system.

Roee9100, 2023-01-06:
6.0.0-kali6-amd64
tp-link t2u plus 
rtl8812au
When im trying to connect some network by wifi the networkmanager cant manage to do it 
its asking me again again the password for the wifi and poping up the wifi authentication login screen 
the networkmanager icon at the upper bar seems to be trying to login again and again and shows me the circle spinning around 
then if i press the icon it stops working and the screen freeze and no reaction to keyboard and mouse
wired connection works fine 

[image]
[image]

Icebice, 2023-01-15:
I'm having this issue as well and only on SPECIFIC wifi access points (so far only one).
I've tried connecting to that wifi access point with my built in wireless network card and I can connect just fine so it's definitely a driver issue.

ZerBea, 2023-01-16:
For reference: Also taints nl80211
`Comm: wireshark Tainted: P        W  O       6.0.0-kali6-amd64`
https://github.com/ZerBea/hcxdumptool/issues/246#issue-1533775961

Roee9100, 2023-01-17:
> For reference: Also taints nl80211 `Comm: wireshark Tainted: P W O 6.0.0-kali6-amd64` [ZerBea/hcxdumptool#246 (comment)](https://github.com/ZerBea/hcxdumptool/issues/246#issue-1533775961)

There is solution?

ZerBea, 2023-01-17:
No, not regarding hcxdumptool/hcxlabtool.

Roee9100, 2023-01-17:
> No, not regarding hcxdumptool/hcxlabtool.

Im not using this tool

ZerBea, 2023-01-17:
We have to wait until the driver will get a fix.
I also only noticed some problems with this driver.

Roee9100, 2023-01-17:
> We have to wait until the driver will get a fix. I also only noticed some problems with this driver.

I plugged the adapter to my second pc(windows) and it couldnt connect to wifi

binbard, 2023-03-01:
I have same problem. Cant connect to any wifi

NayamAmarshe, 2023-06-03:
Same here. The whole system stops responding and the wifi gets stuck on connecting. I can't even poweroff or reboot after that, as no terminal commands work.