aircrack-ng/rtl8812au #1047: install with make on arm64

cdinea, 2023-02-15:
hi i set arch to be arm64/aarch 64 but I run into these issues
install: cannot create regular file '/lib/modules/5.10.65-tegra/kernel/drivers/net/wireless/88XXau.ko': Permission denied
make: *** [Makefile:1739: install] Error 1

KiGamji, 2023-02-17:
try running with sudo

bksieski, 2023-04-24:
I am also having the same issues.  The sudo command does NOT fix this issue.

KiGamji, 2023-04-24:
@bksieski that's in fact either make/distro issue, nothing to do with the driver itself. try running "su" or doing so from the root account

bksieski, 2023-04-24:
i followed the READ.md file for all the commands for rpi and 64 bit. everything works up until that point. i tried sudo but still get the error.  not too familiar with "su"

bksieski, 2023-04-24:
su -i 
then i proceeded to make & make install
no erros.
rebooted
network adapter works
kismet happy with new data source.

@KiGamji thank you for pointing me in the right direction.  appreciate it.