aircrack-ng/rtl8812au #1051: NetworkManager "breaks" driver

IakovlevAA, 2023-03-07:
I've done some tests with AWUS036ACH and get to point that NetworkManager is doing something to adapter and breaking the whole system. I've used airodump-ng to see captured packets and there are difference in driver's work. If NetworkManager is active, before configuring adapter, driver can't capture nothing and switching all channels, even if I stay on 2.4Ghz channels using ```sudo airodump-ng wlan3```.
[image]
As you can see , there are no packets/beacons captured and there are 5Ghz channels.
But it's working fine, if configuring adapter when NetworkManager is down.
[image]
Waiting some time is not a solution, the only thing that helped is ```sudo service NetworkManager stop``` before configuring adapter.

IakovlevAA, 2023-03-07:
Also, there is no problem in OpenWrt, as it doesn't have NetworkManager
[image]

NayamAmarshe, 2023-06-04:
Did you find a solution?

IakovlevAA, 2023-06-05:
I've added commands to configure adapter in rc.local with disabling networkManager